thingsboard/thingsboard-gateway #1777: [BUG] Timeseries not saved to Latest Telemetry—only attributes are stored

Benziza, 2025-05-12:
## Describe the bug  
When publishing MQTT messages through ThingsBoard IoT Gateway with dynamic device provisioning enabled, the gateway auto-creates the target device and parses the JSON correctly—but all numeric fields defined under **timeseries** (e.g. `temperature`) end up in the device’s **Client attributes**. Nothing ever appears in **Latest Telemetry**, despite logs showing `TelemetryEntry` values.

---

## Your Server Environment  
- **Own setup**  
  - Deployment: monolith  
  - Deployment type: docker-compose  
  - ThingsBoard Version: 3.7-stable (`thingsboard/tb-postgres:3.7-stable`)  
  - Edition: Community  
  - OS Name and Version: Ubuntu 22.04 LTS  

### Gateway & Broker Containers  
| Container      | Image                                | Port Mapping  |
|----------------|--------------------------------------|---------------|
| `mqtt-broker`  | `eclipse-mosquitto:2.0`              | `1884:1883`   |
| `tb-gateway`   | `thingsboard/tb-gateway:3.7-stable`  | `5000:5000`   |

---

## Your Client Environment  
**Desktop**  
- OS: Windows 11  
- Browser: Microsoft Edge  
- Version: 136.0.3240.64 (64-bit)

---

## Your Device  
- Connectivity: Gateway (MQTT)  
- ThingsBoard IoT Gateway version: 3.7-stable (`thingsboard/tb-gateway:3.7-stable`)  
- MQTT broker: Mosquitto (`eclipse-mosquitto:2.0`)

---

## To Reproduce  
1. **docker-compose.yml**:
   ```yaml
   services:
     mqtt-broker:
       image: eclipse-mosquitto:2.0
       container_name: mqtt-broker
       ports:
         - "1884:1883"
     tb-gateway:
       image: thingsboard/tb-gateway:3.7-stable
       container_name: tb-gateway
       ports:
         - "5000:5000"
       environment:
         - host=host.docker.internal
         - port=1884
         - accessToken=<GATEWAY_TOKEN>
       depends_on:
         - mqtt-broker
   ```  
2. In ThingsBoard **Gateways → Your Gateway → Configuration → MQTT Connector → Data mapping**, use:
   ```json
   {
      "topicFilter": "rfid/temp",
      "subscriptionQos": 0,
      "converter": {
        "type": "json",
        "deviceInfo": {
          "deviceNameExpression": "TestTemp",
          "deviceNameExpressionSource": "message",
          "deviceProfileExpressionSource": "message",
          "deviceProfileExpression": "default"
        },
        "attributes": [
          {
            "key": "timestamp",
            "type": "string",
            "value": "${timestamp}"
          }
        ],
        "timeseries": [
          {
            "key": "temperature",
            "type": "double",
            "value": "${temperature}",
            "reportStrategy": {
              "type": "ON_RECEIVED"
            }
          }
        ]
      }
    },
   ```  
3. Publish a test message:
   ```bash
   mosquitto_pub -h localhost -p 1884 -t rfid/temp -m '{"rfid":"enter","temperature":55,"timestamp":"2025-05-10T18:10:00Z"}'
   ```  
4. Check ThingsBoard UI under **Devices → Temp_1**:
   - **Client attributes** shows `timestamp: 2025-05-10T18:10:00Z`  
   - **Latest Telemetry** is empty

---

## Expected behavior  
Fields defined under **timeseries** (`temperature`) should be stored as telemetry entries and visible in the **Latest Telemetry** tab of the auto-provisioned device.

Benziza, 2025-05-15:
Hi @imbeacon,

Sorry for the confusion. Just to clarify: attributes are correctly saved under **Client attributes**, but timeseries fields never appear in **Latest Telemetry**.

imbeacon, 2025-05-16:
Hi @Benziza,

I have tried your configuration for connector and it works well:

[image]

I'm a little bit confused with your docker compose file, it looks like you are forwarding port 1884 to mosquitto broker, but at the same time you are using  `     - port=1884` for the gateway service. Could you try to setup ThingsBoard MQTT on 1883 and change `      - port=1884` to `      - port=1883`.

Also, I should mention that if you want to parse ts from timestamp field, with format, different from unix timestamp, you can use `tsField` configuration parameter, it was added in 3.7.4 release:

[image]

For example configuration for your message is:
```

        "timeseries": [
          {
            "key": "temperature",
            "type": "double",
            "value": "${temperature}",
            "reportStrategy": {
              "type": "ON_RECEIVED"
            },
            "tsField": "timestamp"
          }
        ]

```

As you can see, in my case - telemetry was successfully saved, so probably the reason for the issue in your environment or configuration, or MQTT converter tries to parse the value of timestamp as timestamp, but cannot, because it is a datetime string.
My recommendations:
1. Update the image (We released 3.7.4 as 3.7-stable image yesterday)
2. Move ThingsBoard to 1883 and connect to it, avoid usages of the same port
3. Set `tsField` to parameter for telemetry key to `timestamp` - it will automatically parse this datetime string.

If nothing will help - please enable debug logging level for your mqtt connector and for service and provide them after running and sending message for further investigation.

Benziza, 2025-05-18:
Hi @imbeacon 

Thanks for the tips—updating to the thingsboard/tb-edge:3.7-stable image (3.7.4) fixed the timestamp parsing issue, so the problem was in the previous 3.7-stable version.

closed 2025-05-21 by imbeacon