thingsboard/thingsboard-gateway #1778: [HELP] Empty device list

ziotibia81, 2025-05-12:
I'm running one instance of thingsboard-gateway in docker.
This instace is rurring one mqtt connector for one device.

All is working fine, I'm getting all data from mqtt associated to the device.

Thingsboard Gateways page desn't show any device.

Sure I'm missing something.

[image]

**Connector name (If you need help with some connector/converter):**
MQTT Connector

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04.2 LTS
 - Thingsboard IoT Gateway version 3.7.4
 - Thingsboard version 4.0.1

imbeacon, 2025-05-19:
Hi @ziotibia81,

It appears that the devices were already created before the gateway connected. When a gateway sends a connect request for a device, and ThingsBoard doesn’t find an existing device with the same name, it automatically creates the device and establishes a relation between the gateway and the newly created device.

If the devices already exist in the system and you want them to appear under the gateway, you’ll need to manually create the relations between the gateway device and each existing device.

In the future, we plan to revise this logic to rely on the lastConnectedGateway field from the device’s additionalInfo instead of using relations for display purposes. However, at the moment, we don’t have a front-end developer available, so I can’t provide an estimate for when this change will be implemented.

ziotibia81, 2025-05-19:
Thank you for the explanation. The situation is much clearer now.

closed 2025-05-19 by ziotibia81