thingsboard/thingsboard-gateway #1779: [HELP]

ziotibia81, 2025-05-12:
I have an electrical measuring device that exports a list of measurements via MQTT. The topic is formatted as follows: `measuring/[site]/[param]`, and the data is published in plain text.

For example:

```
measuring/site1/L1-Voltage -> 230.0
measuring/site1/Freq -> 50.0
measuring/site1/L1-Current -> 123.4
```
I'm using ThingsBoard IoT Gateway (version 3.7.0) with ThingsBoard version 3.9.1. I've configured an MQTT connector that connects to the broker of the measuring device.

In the _"Data Mapping"_ section, I used a topic filter like `measuring/site1/+` to subscribe to all parameters for `site1`. I set the Payload type to `Bytes` and the value expression to `[:]`. However, I can't figure out how to extract the `key` (e.g., L1-Voltage, Freq) from the MQTT topic using a regular expression or placeholder. If I enter a regular expression into the `key` field, the telemetry data arrives in ThingsBoard with the literal regex string as the key name, not the extracted parameter name.

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04.2 LTS
 - Thingsboard IoT Gateway version 3.7.0
 - Thingsboard version 4.0.1

ziotibia81, 2025-11-05:
Hi! 
I was using Node-RED to translate MQTT messages into the format expected by the ThingsBoard Gateway. 
I am confident that a ThingsBoard Gateway update will allow me to remove the Node-RED layer.

samson0v, 2025-11-06:
Hi @ziotibia81, @MelvinWarnet, thanks for your interest in ThingsBoard IoT Gateway!
We are happy to announce that we will implement this feature in the Gateway v.3.8.1.
So, let's stay connected!

closed 2025-12-09 by samson0v