thingsboard/thingsboard-gateway #1785: [BUG]

granaderainer, 2025-05-16:
**Describe the bug**
Gateway is fetching data from all BACnet Controllers in the network and is mapping this data to the only configured BACnet device in the config.

**Network setup:**
- 4 BACnet controllers

| IP | objectIdentifier |
| --- | --- |
| 192.168.139.23 | 101 |
| 192.168.139.24 | 102 |
| 192.168.139.25 | 103 |
| 192.168.139.26 | 104 |

- thingsboard-gateway
   - 192.168.139.131
   - has configured following device 
     192.168.139.26 - 104

**Issue:**
It seems the gateway receive data from all controllers and uplink all this data to its configured device

For example, Analog Input 3 (see file log.log) is being reported four times with different values at almost the same time (these values are coming from the different BACnet Devices in the network)

[image]

**Config:**

[config.json](https://github.com/user-attachments/files/20251483/config.json)

**Connector name (If bug in the some connector):**
BACnet-Connector

**Error traceback (If available):**

[log.log](https://github.com/user-attachments/files/20251375/log.log)

**Versions (please complete the following information):**
 - OS:  Docker 3.7.4
 - Thingsboard IoT Gateway version 3.7.4
 - Python version 3.11

samson0v, 2025-05-19:
Hi @granaderainer, thanks for your interest in ThingsBoard IoT Gateway!
It seems that it is a bug, we will fix it in the nearest future and let you know.

samson0v, 2025-05-19:
@granaderainer, this bug was fixed. Update the gateway to the latest version using the master branch and let us know about the result.

BastiJoe, 2025-05-19:
we tried the new code.
although only one device is specified (by IP address) multiple destinations receive the readPropertyRequest from the gateway:

[image]

granaderainer, 2025-05-20:
@samson0v the issue still exists.
As @BastiJoe wrote it seems the gateway  starts also collecting data of other BACnet controller as soon as they are available and sending a  i-Am Broadcast message.
After a restart of the gateway it is collecting data of just the configured BACnet Controller again until, the second one sends a i-Am Broadcast (e.g. after restart) 

[image]

samson0v, 2025-05-20:
@granaderainer, the issue was fixed via [commit](https://github.com/thingsboard/thingsboard-gateway/commit/68cbf08fb92bab496008776ae8478722509d1c82).
Please, try to use the gateway from the master branch and let us know about the result.

granaderainer, 2025-05-20:
@samson0v tried it with the  current master branch, but still the same.

[image]

[image]

imbeacon, 2025-05-21:
Hi @granaderainer,

Your screenshot - shows different commit, please try to do pull from original thingsboard-gateway repository. Required commit hash is - 68cbf08fb92bab496008776ae8478722509d1c82

closed 2025-05-21 by imbeacon