thingsboard/thingsboard-gateway #1788: [BUG] General BACNET concept discussion

BastiJoe, 2025-05-19:
**Describe the bug**
We are stumbling over some BACNET Connector bugs and configuration questions:
https://github.com/thingsboard/thingsboard-gateway/issues/1785
https://github.com/thingsboard/thingsboard-gateway/issues/1784
In this context i wanted to share and discuss some general conceptual ideas of BACNET Connector implementation.

We have been dealing with these devices so far for BACNET  - MQTT conversion and made some experience in the BACNET context which we would offer to share.
https://www.mbs-solutions.de/maxi-mk-ii-x-link

From what i understood of the BACNET protocol, the configuration of datapoints is based on device instances (device object instances) rather than on IP adresses.

I would rather see a config to be like this:

[image]

For auto discovery of the devices, it is useful to find the available BACNET devices in the network, but the available  timeseries differ from device to device.
I saw the tb gateway creating multiple devices based in the "deviceNameExpression": ${objectName} which assumes all these BACNET devices provide the same timeserieses on the same bacnet object instances.

Looking forward to your feedback. I am open to any kind of working session. We can provide a bacpypes3 based bacnet emulator for testing as well.

Best regards
Sebastian

**Versions (please complete the following information):**
OS: Docker 3.7.4
Thingsboard IoT Gateway version 3.7.4
Python version 3.11

imbeacon, 2025-05-20:
Hi @BastiJoe,

Thank you for your interest in ThingsBoard IoT Gateway.
Issues that you mentioned - were fixed and fixes are available in the master branch, also they will be added to the next release.
Due to other propositions:
1. deviceNameExpression field variables - we have the following variables, that are available to place in device name - `address`, `objectId`, `vendorId`, `objectName`. You can place them as expressions to deviceNameExpression field and create a name for discovered devices. These are variables that we receive in IAm response and use during device creation. 

2. Automatic data discovering - we have a feature that allows to automatically read all available data from the device, using returned object list from device. It is in beta. To try it - you just need to set your timeseries or attributes to `"*"`, like:
```
...
"timeseries":"*"
...
```
In this case the gateway will try to discover objects and properties and read `present value` from them (In the future we are going to add abilities to read other properties instead/beside with `present value`.

imbeacon, 2025-05-20:
3. `objectIdentifier` - we will discuss this point internally, but at the moment - main filter is address, for example, you can set field `address` for device to `"*:*"` to use global broadcast, or to somethings like `"3001:*"` - for MSTP devices behind router, if router set to 3001, or `"*"`, to discover devices in the local network. Patterns for IP like 192.168.1.XX are also allowed.

BastiJoe, 2025-05-20:
thank you very much for clarification.
When using BACNET we are usually working with EDE Scans (created from BACeye oder YABE). From this EDE Scan we identify the data we want to pull from the network. Usually the EDE Scan only offers device instance, not IP adresses. We have networks with ten thousands of datapoints which would collapse on wildcard polling.
On the other hand we would need to spend extra effort to find the IP adresses for each controller, as we don't have them in EDE file.
Would be super nice to have the option to specify device instance and datapoints object instances.
btw:
is it possible to include the gateway name as variabe into the devicename? as a prefix for example? We are automatizing config generation and that would be super helpful to prefix the device name with the gateway name
`"deviceNameExpression": "${gatewayName}_${objectId}`.

samson0v, 2025-05-20:
Hi @BastiJoe, unfortunately, you cannot use the gateway name in the BACnet device name.

imbeacon, 2025-05-20:
@BastiJoe,

At the moment, as mentioned @samson0v, ability to use gateway name as a name for connected through it device is not present. We will investigate this point.

If routers allow us to get information about connected devices through them - we will be able to send it as attribute for router device and then it will be possible to create relations between routers and devices on ThingsBoard in rule chains to simplify identification of devices and related routers. It will be possible if routers return us this information. 

We will let you know about the results.

imbeacon, 2025-05-21:
Hi @BastiJoe,

We researched ability that we talked about and find it, it is possible to determine router name, address, etc. that is used to connect the device. (Please confirm that we correctly understand that you mean BACnet IP to BACnet MSTP router under "gateway", or please provide a link to example of the gateway you using). 
We have added the task to add this ability to our backlog, we are going to add this ability in the nearest releases.

BastiJoe, 2025-05-21:
Hi @imbeacon 
"deviceNameExpression": "${gatewayName}_${objectId}.
With this i meant the thingsboard-gateway name. I am referring to BACNET IP so far only.
Would be nice to have a thingsboard-gateway name that identifies the site and concat the bacnet controlelr name to identifiy the bacnet controller at this site.
This concatenation with gateway name would be nice to have in any kind of connector.

Regarding point 3. i am attaching a sample EDE File. When we setp up a thingsbaord-gateway on an edge device to collect data from local Bacnet controllers this EDE File is our basis to create a connector configuration. This approach is best practice.
As mentioned, in this EDE File we do not have IP adresses, but we rely on device instances.

[2025-05-19_EDE copy.CSV](https://github.com/user-attachments/files/20361826/2025-05-19_EDE.copy.CSV)

imbeacon, 2025-05-21:
@BastiJoe,

But the gateway that is connected device to can be determined from lastConnectedGateway field in advanced information of the device or using relation from gateway to the device. It is already existing functionality in ThingsBoard and for example existing Gateways UI uses this to display connected through the gateway devices. It will be more correct way to identify the gateway that connects some device.

[image]

We will take a look at csv and return back with results and propositions.

BastiJoe, 2025-05-21:
Thanks @imbeacon for the hint. 
We generate the connector config with a python script, based on the EDE File. This EDE File is not aware of the gateway name and the relation. Therefore it would be useful in automated config generation to be able to use a variable (gateway name) to config the device name that will be created.
example
We have Gateways

- Bacnet-GW-Cologne
- Bacnet-GW-Berlin

with an expression like this we could automatically register the devies

[image]

- Bacnet-GW-Cologne_Bacnet-CTRL-103
- Bacnet-GW-Cologne_Bacnet-CTRL-104
- Bacnet-GW-Berlin_Bacnet-CTRL-101
- Bacnet-GW-Berlin_Bacnet-CTRL-102

imbeacon, 2025-05-21:
@BastiJoe Unfortunately, it is not possible, from the gateway side, because the gateway doesn't know it's name on the platform.

I thought it may be done in rule chain, but not sure that something like the following will work, need to be tested:
device message received -> before saving data you can retrieve entity field (entityName) from related device (relation - Created) -> replace placeholder in deviceName field in metadata -> save data.

Whenever unfortunately from the gateway we cannot get the name or device profile, assigned to the gateway to use them in device names.
I will discuss with product team ability to add ${gatewayName} and ${gatewayProfile} as service placeholders for devices, created through the gateway.

BastiJoe, 2025-05-21:
ah okay, got it. Thank you for clarification and following up. BR

closed 2025-07-03 by imbeacon