thingsboard/thingsboard-gateway #1791: Need help collecting Bacnet data points from Yabe Room Simulator

ttrading, 2025-05-21:
**Describe the issue**
I setup a small lab where I have a Windows machine running the Yabe bacnet simulator, and another virtual machine running the gateway (**v3.4.7**) on Ubuntu. Please note that it is not deployed using docker. I used the deb package to install the gateway. I am using the thingsboard cloud for now to test. Both gateway and the simulator on the Windows machine are using port 47808. 
I see that the gateway sends the "Who-Is" message to the specific bacnet device IP (Windows machine - 10.10.10.200) and the Windows machine sends an "I-am device" response to the local subnet address (10.10.10.255) and the process is stuck there. This message exchange keeps repeating with no other progress.
I tried to use wildcard mask (*) in the IP field for the device, and that did not change the situation. I tried to use wildcard (*) for the application, and that gave errors in the logs due to unrecognized IP.
Please note that I am able to discover points on this simulator using another 3rd party bacnet client.

**Configuration** (Attach your configuration file)
General and connector JSON configurations are attached.

**Connector name (If you need help with some connector/converter):**
Bacnet connection is called Montreal

[connector_configuration.json](https://github.com/user-attachments/files/20360931/connector_configuration.json)
[general_configuration.json](https://github.com/user-attachments/files/20360932/general_configuration.json)

**Error traceback (If it was raised):**
No errors in the logs. The gateway simply cannot pass the discovery stage.

**Versions (please complete the following information):**
 - OS: Ubuntu 22.04.5 LTS
 - Thingsboard IoT Gateway version: v3.4.7
 - Python version: Python 3.10.12

ttrading, 2025-05-21:
Attached is the wireshark output from the Windows machine that hosts the simulator.

[image]

imbeacon, 2025-05-21:
Hi @ttrading,

Thank you for your interest in ThingsBoard IoT Gateway.
It looks like the gateway cannot receive an i-Am message, sent to broadcast.
Could you check and confirm that you can read the device from ubuntu machine with the gateway.
Also, please try to stop the gateway:
`sudo systemctl stop thingsboard-gateway`
And check does port 47808 is not used by any other application?
`sudo netstat -tunlp | grep 47808`
Please confirm that nothing uses this port except the gateway.

Restart the gateway
`sudo systemctl start thingsboard-gateway`

For testing purposes you can set host for device to "*", port to "*" and "timeseries":"*"

Using this configuration - the gateway should send global broadcast and read all objects (present value property) for devices who responded with i-Am.

ttrading, 2025-05-21:
I did this first without changing anything:
```
root@meet:~# sudo systemctl stop thingsboard-gateway
root@meet:~# sudo netstat -tunlp | grep 47808
root@meet:~# sudo systemctl start thingsboard-gateway
root@meet:~# sudo netstat -tunlp | grep 47808
root@meet:~# sudo netstat -tunlp | grep 47808
udp        0      0 10.10.10.255:47808      0.0.0.0:*                           425518/python3
udp        0      0 10.10.10.160:47808      0.0.0.0:*                           425518/python3
root@meet:~# sudo netstat -tunlp | grep 47808
udp        0      0 10.10.10.255:47808      0.0.0.0:*                           425518/python3
udp        0      0 10.10.10.160:47808      0.0.0.0:*                           425518/python3
```

Next I updated the config file for the connector manually under the Advanced menu. Next config is attached. 

[connector_configuration_updated.json](https://github.com/user-attachments/files/20370409/connector_configuration_updated.json)

However, I get the following error: 	
`[bacnet_connector.py] - bacnet_connector __discover_devices - 211 - Error discovering device :: unrecognized format`

About your request: **Could you check and confirm that you can read the device from ubuntu machine with the gateway.**
I can explore the points on 10.10.10.200 machine from another Windows machine in the network but I don't know how to run a bacnet discovery from a linux command line on 10.10.10.160 machine.

ttrading, 2025-05-21:
Please note that I tried to 0.0.0.0 and 255.255.255.255 in the device IP section as well. I am getting the same Who-Is and I-am message exchange and that's it.

samson0v, 2025-05-22:
Hi @ttrading, please, try to use the following config and let us know about the results:
```json
{
  "logLevel": "DEBUG",
  "application": {
    "objectName": "TB_gateway",
    "host": "10.10.10.160",
    "port": "47808",
    "mask": "24",
    "objectIdentifier": 599,
    "vendorIdentifier": 15,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "networkNumber": 0,
    "deviceDiscoveryTimeoutInSec": 5
  },
  "devices": [
    {
      "reportStrategy": {
        "type": "ON_RECEIVED"
      },
      "host": "*",
      "port": "*",
      "deviceInfo": {
        "deviceNameExpression": "BACnet Device ${objectId}",
        "deviceProfileExpression": "default",
        "deviceNameExpressionSource": "expression",
        "deviceProfileExpressionSource": "constant"
      },
      "pollPeriod": 10000,
      "timeseries": "*",
      "attributes": [],
      "attributeUpdates": [],
      "serverSideRpc": []
    }
  ]
}
```

samson0v, 2025-05-22:
@ttrading send also Wireshark dump (or screenshot) of the gateway global broadcast request and responses if it is.

ttrading, 2025-05-22:
I applied the configuration but not seeing any change:

[image]

[image]

[image]

[image]

ttrading, 2025-05-22:
No matter what I change, the connector log reads:
2025-05-22 17:50:01.955 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 209 - WhoIs request sent to device 10.10.10.200:47808
2025-05-22 17:50:22.004 - |INFO| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 198 - Discovering devices...
2025-05-22 17:50:32.007 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 209 - WhoIs request sent to device 10.10.10.200:47808
2025-05-22 17:50:52.068 - |INFO| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 198 - Discovering devices...
2025-05-22 17:51:02.070 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 209 - WhoIs request sent to device 10.10.10.200:47808
2025-05-22 17:51:22.133 - |INFO| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 198 - Discovering devices...
2025-05-22 17:51:32.134 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - __discover_devices - 209 - WhoIs request sent to device 10.10.10.200:47808

I think there is a problem with accepting bacnet messages destined for the local broadcast address

samson0v, 2025-06-23:
Hi @ttrading, unfortunately, broadcast messages will not be received correctly if you are using the gateway from a virtual machine on Windows. You can install the gateway natively on Windows using the pip installation method (or from source using master or release branch). Here is the official guide on how to do it: https://thingsboard.io/docs/iot-gateway/install/pip-installation/
Please let us know if the option above works for you.

closed 2025-07-08 by imbeacon