thingsboard/thingsboard-gateway #1795: [BUG] Upgrade of docker container from 3.6.3 to 3.7.x results in error and frozen TB-GW

JimmyPedersen, 2025-05-21:
**Describe the bug**
I have a docker running v.3.6.3 

[config.zip](https://github.com/user-attachments/files/20370495/config.zip)

on ARM64 HW. It was installed from inside portainer using the thingsboard/tb-gateway at the time when 3.6.3 was the latest version.
It has been running since then with multiple MODBUS connectors (With multiple devices in each connector).
It's stable most of the time but sometimes needs a restart (Currently done with RPC command from button in TB dashboard, works fine). The frequency of hangs has increased as the MODBUS slaves get more use and thus generate more data.
Now we want to upgrade to the latest 3.7 series in the hopes it's more stable.

If I add another docker compose with the 3.7 image it starts fine and I can then add a new connector (even if I use advanced and copy one of the connectors from the 3.6 instance, that also works.
But if I upgrade the 3.6 instance to :latest (ie. use the exact general and connectors configuration. I've added the complete configuration volume content as an attached ZIP file) it doesn't work as it should, doesn't respond to the PING RPC command, and keeps outputting the following log message:

TimeoutError
2025-05-21 14:29:20 - |ERROR| - [base_events.py] - base_events - default_exception_handler - 1785 - Future exception was never retrieved
future: <Future finished exception=TimeoutError()>
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 500, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/thingsboard_gateway/connectors/modbus/slave.py", line 277, in __write
    result = await self.available_functions[function_code](address=address, value=value,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/thingsboard_gateway/connectors/modbus/entities/master.py", line 43, in wrapper
    resp = await func(master, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/thingsboard_gateway/connectors/modbus/entities/master.py", line 112, in write_register
    result = await self.__client.write_register(address=address, value=value, slave=unit_id) # noqa
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.local/lib/python3.11/site-packages/pymodbus/client/base.py", line 292, in execute
    resp = await asyncio.wait_for(req, timeout=self.params.timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 502, in wait_for
    raise exceptions.TimeoutError() from exc

**Versions (please complete the following information):**
 - OS:
    Distributor ID: Debian
    Description:    Debian GNU/Linux 12 (bookworm)
    Release:        12
    Codename:       bookworm

Kernel: Linux tb-iot-gateway 6.6.31-rpi-2712 #3 SMP PREEMPT Tue Jul 30 04:37:42 UTC 2024 
 - Thingsboard IoT Gateway version 3.7 (From 3.6)
 - Python version 3.7.12 (From 3.7.11)

imbeacon, 2025-05-21:
Hi @JimmyPedersen,

Thank you for your interest in ThingsBoard IoT Gateway, could you set service and some of connector logs level to DEBUG level, restart the gateway and send to us logs?

You can collect logs using `docker logs -n 2000 tb-gateway > tb-gateway.log`

JimmyPedersen, 2025-05-21:
Hi.
This is what the logs look like when debug is set on one of each device type of the connectors, and in general settings the Service and Connector.

Log:
[TB-GW.log](https://github.com/user-attachments/files/20371518/TB-GW.log)

It seems the data is received and stored but not sent to TB?
I've double checked that no device is updated and that RPCs from the Gateways dashboard (Like ping) doesn't work.
If Connector settings are changed, they are not synced.

imbeacon, 2025-05-22:
Hi @JimmyPedersen,

It looks like the error appears during reading one device: `Failed to poll NOx_NOT_Ctrl device` but it should not affect data sending to the platform. 

Could you enable DEBUG for connection logs and send us logs? Because from this logs it looks like it cannot connect to single device, but from other data was successfully retrieved and put to storage to be sent to the platform. Maybe incoming RPC locks client and it doesn't send data, but to confirm this - we need at least see connection logs.

JimmyPedersen, 2025-05-22:
Hi @imbeacon 
I've enabled DEBUG on all logs in 'General configuration'.
Here are the logs: 

[_tb-gateway_logs (9).txt](https://github.com/user-attachments/files/20382152/_tb-gateway_logs.9.txt)

In the environment we're communicating it's quite often various devices are offline and not responding.
I expect that to be handled gracefully by TG-GW.

The RPC command (ping) is just something I do manually after a while, after confirming no data is being sent to TB, to confirm that the TB-GW isn't working properly.

closed 2025-05-23 by imbeacon

MrZANE42, 2025-05-23:
Did the failed connection of a modbus slave really lock up the sending of data to thingsboard?
That's rough as unresponsive modbus slaves aren't that unusual.
While there be a new version relased soon? Hope so as this is quite serious.

samson0v, 2025-05-23:
@MrZANE42 @JimmyPedersen , release is planned on 2, June.
Until then, you can use the version from the master branch.

MrZANE42, 2025-05-23:
Is there any way to use the master from Docker (Without creating a custom image)?
If not I'll wait untill the next release

samson0v, 2025-05-23:
@MrZANE42, unfortunately, it is impossible, but you can simply build docker image in 4 steps:
1. git clone https://github.com/thingsboard/thingsboard-gateway.git
2. cd thingsboard-gateway
3. cp docker/Dockerfile .
4. docker build -t tb-gateway .

Docker image built.