thingsboard/thingsboard-gateway #1797: [BUG]  ValueError: Unsupported callback API version: version 2.0 added a callback_api_version,

zlwjzf, 2025-05-22:
version：V3.2
Incorrect description during installation：
tb-mqtt-client library not found - installing...
Collecting tb-mqtt-client
  Downloading tb_mqtt_client-1.13.5.tar.gz (29 kB)
  Preparing metadata (setup.py) ... done
Collecting tb-paho-mqtt-client>=2.1.2 (from tb-mqtt-client)
  Downloading tb_paho_mqtt_client-2.1.2-py3-none-any.whl.metadata (23 kB)
Requirement already satisfied: requests>=2.31.0 in /usr/local/lib/python3.9/site-packages (from tb-mqtt-client) (2.32.3)
Collecting orjson (from tb-mqtt-client)
  Downloading orjson-3.10.18-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (41 kB)
Requirement already satisfied: charset-normalizer<4,>=2 in /usr/local/lib/python3.9/site-packages (from requests>=2.31.0->tb-mqtt-client) (3.4.2)
Requirement already satisfied: idna<4,>=2.5 in /usr/local/lib/python3.9/site-packages (from requests>=2.31.0->tb-mqtt-client) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in /usr/local/lib/python3.9/site-packages (from requests>=2.31.0->tb-mqtt-client) (2.4.0)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.9/site-packages (from requests>=2.31.0->tb-mqtt-client) (2025.4.26)
Downloading tb_paho_mqtt_client-2.1.2-py3-none-any.whl (67 kB)
Downloading orjson-3.10.18-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (132 kB)
Building wheels for collected packages: tb-mqtt-client
  DEPRECATION: Building 'tb-mqtt-client' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'tb-mqtt-client'. Discussion can be found at https://github.com/pypa/pip/issues/6334
  Building wheel for tb-mqtt-client (setup.py) ... done
  Created wheel for tb-mqtt-client: filename=tb_mqtt_client-1.13.5-py3-none-any.whl size=30372 sha256=81433eab0a9cca0538b06aa3bb01d67178d29d01caaf4f12dafe28d751331776
  Stored in directory: /root/.cache/pip/wheels/a8/92/84/3a4b455a345c58969171bda107f93f62f75e2e009163e518a7
Successfully built tb-mqtt-client
Installing collected packages: tb-paho-mqtt-client, orjson, tb-mqtt-client
Successfully installed orjson-3.10.18 tb-mqtt-client-1.13.5 tb-paho-mqtt-client-2.1.2
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user-action option if you know what you are doing and want to suppress this warning.
Collecting mmh3
  Downloading mmh3-5.1.0-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (16 kB)
Downloading mmh3-5.1.0-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (98 kB)
Installing collected packages: mmh3
Successfully installed mmh3-5.1.0
""2025-05-22 06:22:17" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
""2025-05-22 06:22:18" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 0"
""2025-05-22 06:22:18" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 269 - Subscribed to *|* with id 1 for device *"
""2025-05-22 06:22:18" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import GrpcMqttConnector from /thingsboard_gateway/grpc_connectors/mqtt."
""2025-05-22 06:22:18" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import MqttConnector from /thingsboard_gateway/connectors/mqtt."
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted mapping handlers: 3"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected mapping handlers: 0"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted serverSideRpc handlers: 2"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected serverSideRpc handlers: 0"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted connectRequests handlers: 2"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected connectRequests handlers: 0"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted disconnectRequests handlers: 2"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected disconnectRequests handlers: 0"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted attributeRequests handlers: 1"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected attributeRequests handlers: 0"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted attributeUpdates handlers: 1"
""2025-05-22 06:22:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected attributeUpdates handlers: 0"
""2025-05-22 06:22:18" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - _connect_with_connectors - 609 - Unsupported callback API version: version 2.0 added a callback_api_version, see docs/migrations.rst for details"
Traceback (most recent call last):
  File "/thingsboard_gateway/gateway/tb_gateway_service.py", line 599, in _connect_with_connectors
    connector = self._implemented_connectors[connector_type](self,
  File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 157, in __init__
    self._client = Client(client_id, protocol=MQTT_VERSIONS[self._mqtt_version])
  File "/root/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 772, in __init__
    raise ValueError(
ValueError: Unsupported callback API version: version 2.0 added a callback_api_version, see docs/migrations.rst for details

zlwjzf, 2025-05-29:
Hello, after I discovered this bug, I re-established the V3.2 environment in the Docker environment. So in the tb_gateway.yaml file, only the connection information of TB was modified, while all other configurations remained default (there would be a default mqtt.json configuration item named MQTT Broker Connector). The aforementioned issue seems to be caused by using an overly high version of the library when creating the MQTT connection. However, I'm not sure how to set the MQTT version during the installation process.

samson0v, 2026-01-14:
@zlwjzf 

📢 **a little announcement:**
We just launched our [official Discord server](https://discord.com/invite/mJxDjAM3PF), we'd love to see you there!

Prus-Oleksandr, 2026-09-03:
Hi [@zlwjzf](https://github.com/zlwjzf),

Thank you for reporting this. We've re-checked this with the current release of the library, and the issue no longer reproduces: callback_api_version now has a default value (CallbackAPIVersion.VERSION1) in the current paho-mqtt/tb-paho-mqtt-client release, so the connector no longer raises this ValueError — it only emits a deprecation warning. Closing this as resolved upstream.

closed 2026-09-03 by samson0v