thingsboard/thingsboard-gateway #1804: [BUG] An error occurred in the Periodical Reporting Thread

andersenthomas98, 2025-05-26:
**Describe the bug**
This [ValueError](https://github.com/thingsboard/thingsboard-gateway/blob/7a246badfe9d02d0d37010575801b1d1d74cac70/thingsboard_gateway/gateway/report_strategy/report_strategy_service.py#L223) is raised consistently after running the gateway for some time. When this occurs: 
- CPU surges.
- The gateway stops transmitting telemetry to ThingsBoard.
- Repeated error logs are generated.

The error persists until the gateway is restarted.

**Error traceback:**
```
tb-gateway  | 2025-05-26 19:35:26 - |ERROR| - [tb_logger.py] - tb_logger - exception - 182 - An error occurred in the Periodical Reporting Thread: Data record for key 'DatapointKey(key=M502203-28.Stw, report_strategy=None)' is absent in the cache
tb-gateway  | Traceback (most recent call last):
tb-gateway  |   File "/thingsboard_gateway/gateway/report_strategy/report_strategy_service.py", line 222, in __periodical_reporting
tb-gateway  |     raise ValueError(f"Data record for key '{key}' is absent in the cache")
tb-gateway  | ValueError: Data record for key 'DatapointKey(key=M502203-28.Stw, report_strategy=None)' is absent in the cache
tb-gateway  | 2025-05-26 19:35:26 - |ERROR| - [report_strategy_service.py] - report_strategy_service - __periodical_reporting - 267 - Too many errors occurred in the Periodical Reporting Thread for 5 seconds.                            Suppressing further error messages.
tb-gateway  | 2025-05-26 19:35:32 - |ERROR| - [tb_logger.py] - tb_logger - exception - 182 - An error occurred in the Periodical Reporting Thread: Data record for key 'DatapointKey(key=M502203-28.Stw, report_strategy=None)' is absent in the cache
tb-gateway  | Traceback (most recent call last):
tb-gateway  |   File "/thingsboard_gateway/gateway/report_strategy/report_strategy_service.py", line 222, in __periodical_reporting
tb-gateway  |     raise ValueError(f"Data record for key '{key}' is absent in the cache")
tb-gateway  | ValueError: Data record for key 'DatapointKey(key=M502203-28.Stw, report_strategy=None)' is absent in the cache
tb-gateway  | 2025-05-26 19:35:32 - |ERROR| - [report_strategy_service.py] - report_strategy_service - __periodical_reporting - 267 - Too many errors occurred in the Periodical Reporting Thread for 5 seconds.                            Suppressing further error messages.
tb-gateway  | 2025-05-26 19:35:38 - |ERROR| - [tb_logger.py] - tb_logger - exception - 182 - An error occurred in the Periodical Reporting Thread: Data record for key 'DatapointKey(key=M502203-28.Stw, report_strategy=None)' is absent in the cache
tb-gateway  | Traceback (most recent call last):
tb-gateway  |   File "/thingsboard_gateway/gateway/report_strategy/report_strategy_service.py", line 222, in __periodical_reporting
tb-gateway  |     raise ValueError(f"Data record for key '{key}' is absent in the cache")
tb-gateway  | ValueError: Data record for key 'DatapointKey(key=M502203-28.Stw, report_strategy=None)' is absent in the cache
tb-gateway  | 2025-05-26 19:35:38 - |ERROR| - [report_strategy_service.py] - report_strategy_service - __periodical_reporting - 267 - Too many errors occurred in the Periodical Reporting Thread for 5 seconds.                            Suppressing further error messages.
```

**Versions:**
 - OS: Ubuntu 24.04 
 - Docker version 27.3.1
 - Thingsboard IoT Gateway version 3.7-stable

Any help here would be much appreciated.

andersenthomas98, 2025-05-30:
Let me know if there is any more information that would be useful to fix this. I would also be happy to contribute to a fix, however, I would need some guidance on where to start.

imbeacon, 2025-06-03:
Hi @andersenthomas98,

Please try the latest release (3.7.5) you can get it by changing image version in docker compose file to `thingsboard/tb-gateway:tb-latest` the issue should be solved there.

andersenthomas98, 2025-06-03:
Thanks! I will try the latest release

closed 2025-06-23 by imbeacon