thingsboard/thingsboard-gateway #1812: [BUG] PR #1745 breaks use received timestamp in mqtt connector

rroemhild, 2025-06-03:
The `USE_RECEIVED_TS_PARAMETER` flag was introduced in version v3.7.3 via [PR #1718](https://github.com/thingsboard/thingsboard-gateway/pull/1718).
However, its functionality appears to be unintentionally overridden by a change introduced in [PR #1745](https://github.com/thingsboard/thingsboard-gateway/pull/1745).

Specifically, the timestamp is overwritten here:
https://github.com/thingsboard/thingsboard-gateway/blob/6095fc76f5050b37a5622484dd6a939e719c1f88/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py#L119

As a result, setting USE_RECEIVED_TS_PARAMETER to true no longer has the intended effect.

@Andry925 do you have any suggestions for resolving this conflict?

**Versions:**
 - Thingsboard IoT Gateway version v3.7.4, v3.7.5

imbeacon, 2025-06-03:
Hi @rroemhild,

You are right, but now - the gateway is able to parse a much more timestamps, if you don't want to use the added timestamp - you can set use server timestamp to true for save timeseries rule node, it will be more correct way to handle this situation:

[image]

closed 2025-06-03 by imbeacon

imbeacon, 2025-06-04:
Hi @rroemhild,

I see what you mean, please find the following [commit](https://github.com/thingsboard/thingsboard-gateway/commit/34e9a40fd63f7bc056e57176f37757de8ce64c7b), it allows to use the discussed behaviour, but in the future I think, we will adopt a little bit changes, to avoid message splitting.