thingsboard/thingsboard-gateway #1818: [BUG] : Docker commands section is empty after importing widget bundle and dashboard

pmexiajr, 2025-06-05:
Hi,

I'm using ThingsBoard Professional Edition v4.0.1, and I'm facing an issue after following the official documentation.

I completed all the steps in the [Getting Started guide](https://thingsboard.io/docs/iot-gateway/getting-started/#optional-import-gateway-widgets-bundle-and-dashboard), including importing the gateway widget bundle and the dashboard.

However, when I create a new gateway device, the Docker commands section is completely empty — no commands are shown, and there's no option to download or deploy the gateway via Docker from the UI.

Is there any additional step or configuration required to enable this feature on PE 4.0.1?

Thank you in advance!

[image]

imbeacon, 2025-06-19:
Hi @pmexiajr,

Thank you for your interest in ThingsBoard IoT Gateway.

Unfortunately, getting started a little bit outdated, and we are going to update it in the nearest future.
You can find a separated menu point `Gateways` in ThingsBoard, it contains the same UI. Dashboard import is no longer required:

[image]

pmexiajr, 2025-06-23:
Thank you very much!