thingsboard/thingsboard-gateway #1831: [BUG] Gateway v3.7.5 on Windows crashes and disconnects after a successful Modbus data read

dechambrier, 2025-06-18:
**Describe the bug**

The ThingsBoard Gateway (v3.7.5) running on Windows connects successfully to both a Modbus TCP PLC and the ThingsBoard Cloud server. The gateway successfully polls the PLC and receives the correct data values. However, immediately after the first successful data poll, the gateway's connection to the ThingsBoard MQTT broker is severed. The gateway then enters a continuous loop of reconnecting and disconnecting every few seconds.

This issue appears to be a fatal crash/bug in the gateway's core logic that occurs when it tries to process or publish the first valid data packet received from the Modbus connector.

**Steps to Reproduce**

Install ThingsBoard Gateway v3.7.5 on a Windows machine using pip.
Configure the main thingsboard.json (or .yaml) file to connect to a ThingsBoard server (e.g., ThingsBoard Cloud) using an access token.
Configure a Modbus connector to poll a valid timeseries register from a working Modbus TCP device.
Ensure the gateway is configured to read at least one telemetry value that is known to be correct.
Start the gateway.
Expected behavior
The gateway should connect to the PLC, read the data, successfully publish the telemetry to ThingsBoard, and remain connected, continuing to poll at the specified interval.

**Actual behavior**

The gateway starts, connects to the PLC, and connects to the ThingsBoard server. The logs confirm a successful Modbus read. Immediately after this first successful read, the MQTT connection drops with reason code None. The gateway then enters a connect-disconnect loop, making it unusable.

A representative log snippet of the loop:

```
...
|INFO| - [slave.py] - slave - connect - 209 - Connected to [Your PLC Name]
...
|INFO| - [tb_client.py] - tb_client - _on_connect - 313 - MQTT client connected to platform [Your TB Host]
...
// Data is successfully read from PLC at this point
...
|WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 568 - MQTT client was disconnected with reason code None (Description not found.)
...
// Reconnect attempts begin
```

**Environment**

OS: Windows
ThingsBoard IoT Gateway version: 3.7.5
Python version: 3.13
Installation method: pip install thingsboard-gateway
Connector: Modbus TCP
ThingsBoard platform: ThingsBoard Cloud (mqtt.eu.thingsboard.cloud)

**Troubleshooting Steps Performed (to rule out common issues)**

This issue was troubleshot extensively. The following potential causes were eliminated:

- Local/Network Firewall: The issue persists even with the local Windows Firewall completely disabled. The PLC Isolation Test (using an invalid IP) proved the gateway stays connected to ThingsBoard when not talking to the PLC, ruling out a general network block.
- Access Token Conflict: The access token was regenerated, with no change in behavior.
- Keep-Alive/Timeout: The disconnect is instant (<1 second after data read), not a timeout. keepaliveSeconds was configured with no effect.
- Invalid Modbus Address/Data: The user confirmed that the gateway is receiving the correct numerical data from the PLC before it crashes. The issue persists even when simplifying the read from a 32float to a 16uint.
- reportStrategy: The crash occurs both with and without the reportStrategy configured in the main config file.

The logical conclusion is that a bug exists in the gateway's core code that causes a crash when handling a valid, successfully polled data packet from the Modbus connector.

dechambrier, 2025-06-20:
Thanks!

For info, I'm in docker now, and observe the same behaviour.

dechambrier, 2025-06-23:
I systematically tested several hypotheses to find the source of the restart loop.

Network & VPN: Ruled Out. I analyzed your PC's routing table and proved that neither the network path nor Tailscale was the cause.
Polling Speed: Ruled Out. I increased the polling interval to 5 and 10 seconds, but the disconnect loop persisted, proving it wasn't a timing issue.
Server-Side Policies: Ruled Out. I checked the server-side logs in the ThingsBoard UI and confirmed the server was not actively terminating the connection.
Execution Environment: Ruled Out. I confirmed the issue was identical when running on Windows directly and in Docker, proving it wasn't a container-specific problem. I do not have a gateway running in the background. 

The definitive tests were:

Disabling the Connector: When the "Wago Connector" was disabled in the UI, the gateway connection is  completely stable.
Emptying the timeseries List: When the connector was enabled but had no tags to poll, the connection also remained stable.

Conclusion of the Isolation Process:

The problem is triggered specifically and only when the Modbus connector executes a poll for timeseries data. The gateway's core service is stable, but the action of polling for data causes the entire application to perform a clean shutdown, resulting in the connection loop.

Do you still want me to try with wireshark?

dechambrier, 2025-06-24:
@imbeacon I am hitting limits for the device, sorry I missed that, I was not coming back to the homepage much. 

But I don't understand how it's possible, I'm on prototype plan and this is the query: 

`{
  "name": "Modbus_PLC_Connector_1",
  "id": "e0b1c2d3-f4e5-6789-abcd-1234567890ab",
  "master": {
    "slaves": [
      {
        "host": "192.168.X.Y",
        "port": 502,
        "method": "socket",
        "unitId": 1,
        "deviceName": "PLC_System_A",
        "deviceType": "Industrial_Controller",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "LITTLE",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 10000,
        "connectAttemptTimeMs": 500,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 30000,
        "type": "tcp",
        "attributes": [],
        "timeseries": [
          {
            "tag": "levelSensor_A",
            "type": "32float",
            "address": 12304,
            "objectsCount": 2,
            "functionCode": 3
          },
          {
            "tag": "columnLevel_B",
            "type": "32float",
            "address": 12298,
            "objectsCount": 2,
            "functionCode": 3
          },
          {
            "tag": "pumpEnable_C",
            "type": "bits",
            "address": 12291,
            "objectsCount": 1,
            "functionCode": 1,
            "bitTargetType": "bool"
          },
          {
            "tag": "pumpEnable_D",
            "type": "bits",
            "address": 12296,
            "objectsCount": 1,
            "functionCode": 1,
            "bitTargetType": "bool"
          },
          {
            "tag": "phValue_E",
            "type": "32float",
            "address": 12292,
            "objectsCount": 2,
            "functionCode": 3
          },
          {
            "tag": "temperatureSensor_F",
            "type": "32float",
            "address": 12288,
            "objectsCount": 2,
            "functionCode": 3
          },
          {
            "tag": "storageLevel_G",
            "type": "32float",
            "address": 12384,
            "objectsCount": 2,
            "functionCode": 3
          },
          {
            "tag": "tankLevel_H",
            "type": "32float",
            "address": 12352,
            "objectsCount": 2,
            "functionCode": 3
          },
          {
            "tag": "productLevel_I",
            "type": "32float",
            "address": 12392,
            "objectsCount": 2,
            "functionCode": 3
          },
          {
            "tag": "flowRate_J",
            "type": "32float",
            "address": 12310,
            "objectsCount": 2,
            "functionCode": 3
          }
        ],
        "attributeUpdates": [],
        "rpc": []
      }
    ]
  }
}`

imbeacon, 2025-06-24:
This behavior can occur when the gateway is restarted. Unfortunately, ThingsBoard currently does not provide an API to retrieve the number of available tokens for rate limits, which may lead to temporary rate limiting issues after a restart.

In most cases, the issue should stabilize after the gateway runs continuously for about 2 hours. If it does not, a workaround is to stop the gateway, wait for 2 hours (to allow the rate limit windows to reset), and then start it again. Since the maximum rate limit on thingsboard.cloud is typically enforced over a 1-hour window, this approach should help avoid further triggering of the limits.

We are aware of this situation and have already reported it to the ThingsBoard product team. Unfortunately, there is no immediate solution available, but it is on their radar for future improvements.

imbeacon, 2025-06-25:
I mean stop the gateway, not just connector, it is an issue with connection between the gateway and platform with rate limits, unfortunately stopping of the connector won't help, because this doesn't interrupt connection between the gateway and the platform.