thingsboard/thingsboard-gateway #1832: [HELP]gRPC connector

Liuyuan0803, 2025-06-19:
I want to build a gRPC connector for communicating between the thingsboard gateway and devices to getting data from devices and putting these data into thingsboard by gateway. But I found a file holder named _grpc_service_ , it cannot achieve my function. I want to get some suggests to getting my goal. My ideal way is using _grpc_service_.

Environment

OS: Windows
Thingsboard Gateway Version: 3.4.2
Thingsboard Platform Version: 3.6.2

Thanks everyone.

imbeacon, 2025-06-19:
Hi @Liuyuan0803,

Thank you for your interest in ThingsBoard IoT Gateway. 
GRPC service in the gateway was designed to add ability to communicate between connectors and the gateway core using GRPC. It is not relates to data retrieving or communication to devices using it. 
To add functionality like you described, you can add a custom connector, or, if you want - you can create a generic GRPC connector, which will allow to use different proto files and create a PR, we will be happy to add connector like this to the main connectors list.

Liuyuan0803, 2025-06-19:
These files are mentioned in my question. Your meaning is that these files' function is for communicating between thingsboard platform and gateway? Or if I could achieve my goal based on these files?

closed 2025-08-12 by samson0v