thingsboard/thingsboard-gateway #1849: [FEATURE] REST Connector - add support for client SSL certs

apachler, 2025-06-27:
**Describe the feature**

Below change of the `__send_request` function gives following additional possibility:
1. Gives us the possibility to use a client certificate for outgoing requests

```python
    def __send_request(self, request_dict, converter_queue, logger, with_queue=True):
        url = ""
        try:
            request_dict["next_time"] = time() + request_dict["config"].get(
                "scanPeriod", 10
            )

            if str(request_dict["config"]["url"]).lower().startswith("http"):
                url = request_dict["config"]["url"]
            else:
                url = "http://" + request_dict["config"]["url"]

            logger.debug(url)
            security = None

            if (
                request_dict["config"]
                .get("security", {})
                .get("type", "anonymous")
                .lower()
                == "basic"
            ):
                security = HTTPBasicAuthRequest(
                    request_dict["config"]["security"]["username"],
                    request_dict["config"]["security"]["password"],
                )

            cert = None
            if (
                request_dict["config"]
                .get("security", {})
                .get("type", "anonymous")
                .lower()
                == "cert"
            ):
                if request_dict["config"]["security"].get("key", ""):
                    cert = (
                        request_dict["config"]["security"]["cert"],
                        request_dict["config"]["security"]["key"],
                    )
                else:
                    cert = request_dict["config"]["security"]["cert"]

            request_timeout = request_dict["config"].get("timeout")

            data = {"data": request_dict["config"]["data"]}
            params = {
                "method": request_dict["config"].get("HTTPMethod", "GET"),
                "url": url,
                "timeout": request_timeout,
                "allow_redirects": request_dict["config"].get("allowRedirects", False),
                "verify": request_dict["config"].get("SSLVerify"),
                "auth": security,
                "cert": cert,
                **data,
            }
            logger.debug(url)

            if request_dict["config"].get("httpHeaders") is not None:
                params["headers"] = request_dict["config"]["httpHeaders"]

            logger.debug("Request to %s will be sent", url)
            response = None
            data_to_storage = []
            try:
                response = request_dict["request"](**params)

            except Timeout:
                logger.error("Timeout error on request %s.", url)
                data_to_storage.append({"error": "Timeout", "code": 408})
            except RequestException as e:
                logger.error("Cannot connect to %s. Request exception.", url)
                data_to_storage.append({"error": str(e)})
                logger.debug(e)
            except ConnectionError:
                logger.error("Cannot connect to %s. Connection error.", url)
                data_to_storage.append(
                    {"error": f"Cannot connect to target url: {url}"}
                )

            if response and response.ok:
                try:
                    data_to_storage.append(response.json())
                except UnicodeDecodeError:
                    data_to_storage.append(response.content)
                except JSONDecodeError:
                    data_to_storage.append(response.content)

                if (
                    len(data_to_storage) == 3
                    and with_queue
                    and not converter_queue.full()
                ):
                    converter_queue.put(data_to_storage)
                    self.statistics["MessagesReceived"] = (
                        self.statistics["MessagesReceived"] + 1
                    )
            else:
                if response is not None:
                    logger.error(
                        "Request to URL: %s finished with code: %i. Cat information: http://http.cat/%i",
                        url,
                        response.status_code,
                        response.status_code,
                    )
                    logger.debug("Response: %r", response.text)
                    data_to_storage.append(
                        {"error": response.reason, "code": response.status_code}
                    )

                if with_queue:
                    converter_queue.put(data_to_storage)

                self.statistics["MessagesReceived"] = (
                    self.statistics["MessagesReceived"] + 1
                )

            if not with_queue:
                return data_to_storage
        except Exception as e:
            logger.exception(e)
```

```json
    "attributeUpdates": [
      {
        "HTTPMethod": "POST",
        "SSLVerify": false,
        "httpHeaders": {
          "CONTENT-TYPE": "application/json"
        },
        "security": {
          "type": "cert",
          "cert": "/thingsboard_gateway/config/cert.pem",
          "key": "/thingsboard_gateway/config/key.pem"
        },
```
**Connector name (If bug in the some connector):**
REST Connector

**Versions (please complete the following information):**
 - OS: SLES 15
 - Thingsboard IoT Gateway version 3.7.5

imbeacon, 2025-06-30:
Hi @apachler,

Thank you for your participation in the project life. As I mentioned in another issue, could you open a pull request with these changes?

apachler, 2025-06-30:
OK, i am going to create a pull request.