thingsboard/thingsboard-gateway #1851: [FEATURE] Request Connector - add support for different hosts for one connector

apachler, 2025-06-27:
**Describe the feature**

Below change of the `__send_request` function gives following additional possibility:
1. Gives us the possibility to configure different hosts for one connector

```python
    def __send_request(self, request, converter_queue, logger):
        url = ""
        try:
            request["next_time"] = time() + request["config"].get("scanPeriod", 10)
            request_url_from_config = request["config"]["url"]
            request_url_from_config = (
                str("/" + request_url_from_config)
                if not request_url_from_config.startswith("/")
                and not request_url_from_config.startswith("http")
                else request_url_from_config
            )
            logger.debug(
                "Obtained request url from config - %s ", request_url_from_config
            )
            if not request_url_from_config.startswith("http"):
                url = self.__host + request_url_from_config
            else:
                url = request_url_from_config
            request_timeout = request["config"].get("timeout", 1)
            params = {
                "method": request["config"].get("httpMethod", "GET"),
                "url": url,
                "timeout": request_timeout,
                "allow_redirects": request["config"].get("allowRedirects", False),
                "verify": self.__ssl_verify,
                "auth": self.__security,
                "data": request["config"].get("data", {}),
            }
            logger.debug("Full url request has been formed - %s", url)

            if request["config"].get("httpHeaders") is not None:
                params["headers"] = request["config"]["httpHeaders"]
            logger.debug("Request to %s will be sent", url)
            response = request["request"](**params)

            if request.get("withResponse"):
                converter_queue.put(response)
                return

            if response and response.ok:
                if not converter_queue.full():
                    config_converter_data = [url, request["converter"]]
                    try:
                        config_converter_data.append(response.json())
                    except UnicodeDecodeError:
                        config_converter_data.append(response.content)
                    except JSONDecodeError:
                        config_converter_data.append(response.content)

                    if len(config_converter_data) == 3:
                        self.__convert_data(config_converter_data)
            else:
                logger.error(
                    "Request to URL: %s finished with code: %i",
                    url,
                    response.status_code,
                )
        except Timeout:
            logger.error("Timeout error on request %s.", url)
        except RequestException as e:
            logger.error("Cannot connect to %s. Connection error.", url)
            logger.debug(e)
        except ConnectionError:
            logger.error("Cannot connect to %s. Connection error.", url)
        except Exception as e:
            logger.exception(e)
```

```json
{
  "host": "http://",
  "SSLVerify": false,
  "security": {
    "type": "anonymous"
  },
  "mapping": [
    {
      "url": "http://10.217.10.230/cgi-bin/ope/allch.cgi",
      "httpMethod": "GET",
```

**Connector name:**
Request Connector

**Versions (please complete the following information):**
 - OS: SLES 15
 - Thingsboard IoT Gateway version 3.7.5

imbeacon, 2025-06-30:
Hi @apachler,

Thank you for contributing to the project! Would you be open to submitting a pull request with these changes?
Alternatively, if you prefer, we can integrate them on our end - just let us know what works best for you.

apachler, 2025-06-30:
OK, i am going to create a pull request.

apachler, 2025-07-01:
see PR https://github.com/thingsboard/thingsboard-gateway/pull/1856