thingsboard/thingsboard-gateway #1853: [FEATURE] BLE Connector manufacturer_data vs service_data

powermik, 2025-06-28:
I am trying to integrate some plant sensors: 
http://www.huahuacaocao.com/product or search for "hhcc flower care sensor"

They send the measurements as "service_data"
In 
https://github.com/thingsboard/thingsboard-gateway/blob/fac5e58804df4881357e5276fad6280777222e09/thingsboard_gateway/connectors/ble/device.py#L257 it is hardcoded to only get manufacturer_data.

Also it would be nice to have the possibility to send the received advertisement bytestring as Hexvaluestring like '0xDEADBEEF' to Thingsboard for further processing there, or add something like "ValueCode" as a supplement to "ValueExpression" for more complicated calculations in python code form.

imbeacon, 2025-07-01:
Hi @powermik,

Thank you for your interest in ThingsBoard IoT Gateway.
You are right, at the moment the gateway can use only manufacturer_data, from advertisement. We are going to improve BLE connector in one of the following release and improve the reading of the advertising data. 

So, how do you think, which is the better option:

1. Add fields from advertising data to value expression - like `${manufacturerData}`, `${serviceData}`, `${localName}`, `${rssi}` etc.

2. Automatically parse incoming data and send as attributes, with the same names.

Unfortunately we cannot extract raw bytes from the response (maybe we will find the way during development, but at the moment I don't see this ability).

imbeacon, 2025-07-08:
Hi,

The new version is released. Please try to use it, feel free to open new issue if you discover something.

closed 2025-07-08 by imbeacon