thingsboard/thingsboard-gateway #1861: [BUG] Modbus timeout with 3.7.5

majuss, 2025-07-03:
**Describe the bug**
I have a modbus master config which includes connections with several slaves. 3.7.5 introduced a modbus timeout. The line in the config: ""timeout": 35," was also there in 3.7.4 so I'am not sure what is different in .5. Long story short, the following config (its 13k lines) does work on .4 but .5 doesnt work at all, all polling is timing out.

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

```json
{
  "master": {
    "slaves": [
      {
        "type": "serial",
        "method": "rtu",
        "port": "/dev/ttyRS485",
        "baudrate": 115200,
        "timeout": 35,
        "stopbits": 1,
        "bytesize": 8,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 30000,
        "reportStrategy": {
          "type": "ON_CHANGE_OR_REPORT_PERIOD",
          "reportPeriod": 300000
        },
        "unitId": 1,
        "deviceName": "BACnet-13259-135-138667",
        "deviceType": "BACNETController_tbiotgw",
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "attributes": [],
        "timeseries": [
          {
            "tag": "up_BO_15",
            "type": "bits",
            "functionCode": 1,
            "objectsCount": 1,
            "address": 0
          },
          {
            "tag": "up_BO_14",
            "type": "bits",
            "functionCode": 1,
            "objectsCount": 1,
            "address": 1
          },
          {
            "tag": "up_BO_13",
            "type": "bits",
            "functionCode": 1,
            "objectsCount": 1,
            "address": 2
[... 1473 lines cut ...]
            "address": 24
          }
        ],
        "attributeUpdates": []
      },
      {
        "type": "serial",
        "method": "rtu",
        "port": "/dev/ttyRS485",
        "baudrate": 115200,
        "timeout": 35,
        "stopbits": 1,
        "bytesize": 8,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 30000,
        "reportStrategy": {
          "type": "ON_CHANGE_OR_REPORT_PERIOD",
          "reportPeriod": 300000
        },
 ...
```

majuss, 2025-07-04:
I set it to debug and the timeout to 300 seconds. There are now green log lines and it can read registers but the telemetry is still not updated.

closed 2025-07-04 by imbeacon

samson0v, 2025-07-04:
Hi @majuss, try to use the latest version of the gateway and let us know about the result.

samson0v, 2025-07-14:
@majuss, the latest gateway was released, so you can update your gateway via docker hub (or any other official installation)