thingsboard/thingsboard-gateway #1880: [HELP] Missing 3.7.5 and 3.7.6 Tags on Docker Hub

ishikht, 2025-07-11:
Hi,

It appears that Docker Hub is missing the specific tags for versions 3.7.5 and 3.7.6. Currently, only the 3.7-latest tag is available, which makes it unclear which exact version it refers to.

For clarity and reproducibility, could you please publish the version-specific tags (3.7.5, 3.7.6, etc.) to Docker Hub?

Without explicit version tags, it's difficult to guarantee which version is being used, and updates to the 3.7-latest tag can cause unintentional changes in behavior.

Thanks!

imbeacon, 2025-07-11:
Hi @ishikht,

Thank you for your noticing, we will update it in few minutes, 3.7.6 is already available.

ishikht, 2025-07-11:
Thank you @imbeacon !

apachler, 2025-07-14:
tag 3.7.5 would be nice to have ;)