thingsboard/thingsboard-gateway #1885: [HELP] Cannot download Docker compose file following the "getting started" guide.

abIIcd, 2025-07-15:
**Describe the issue**
Following the Getting started with ThingsBoard IoT Gateway guide, I import 2 json files and create a gateway device. But when I clicked "launch common" button, there is nothing.

[image]

I meet the same issue both in my local thingsboard service(in docker) and in Live Demo.

**Versions (please complete the following information):**
 - OS: Ubuntu 22.04.5 LTS
 - Thingsboard community edtion, version 4.0.2

samson0v, 2025-07-15:
Hi @abIIcd, thanks for your interest in ThingsBoard IoT Gateway!
You don't need to import the dashboard anymore from ThingsBoard v.3.9.0.
You can find gateway management on the left sidebar under the Entities tab (as on the screenshot below)

[image]

abIIcd, 2025-07-15:
That fixed it, thanks! You are so efficient.

closed 2025-07-15 by abIIcd