thingsboard/thingsboard-gateway #1888: [BUG] OPC path addressing for attributes_updates not working

BastiJoe, 2025-07-15:
**Describe the bug**
When i use path addressing for OPC UA attributes_updates i get errors.
For timeseries the same path addressing works.
When i use the identifier addressing for attributes_updates  it also works.

[image]

**Connector name (If bug in the some connector):**
OPC-UA Connector

**Error traceback (If available):**
[opcua_connector.py] - opcua_connector on_attributes_update - 1093 - Error during processing shared attribute update: 'NoneType' object has no attribute 'write_value'
[opcua_connector.py] - opcua_connector __write_value - 1282 - Can not find node for provided path None
[opcua_connector.py] - opcua_connector find_full_node_path - 1057 - Error during node lookup for 'AV\\.0': Connection is closed

**Versions (please complete the following information):**
 - OS: docker
 - Thingsboard IoT Gateway version 3.7.6

samson0v, 2025-07-16:
Hi @BastiJoe, thanks for your interest in ThingsBoard IoT Gateway!
You should use the following configuration format for attribute updates:

```json
"attributes_updates": [
  {
    "key": "nodeById",
    "type": "path",
    "value": "ns=2;i=23"
  },
  {
    "key": "nodeByAbsolutePath",
    "type": "path",
    "value": "Root\\.Objects\\.MyObject\\.MyStringVariable"
  }
]
```

BastiJoe, 2025-07-16:
Hi @samson0v 
i tried this:

[image]
nd still get this error:
[image]

as reference, this works:

[image]