thingsboard/thingsboard-gateway #1890: [HELP] How to get temperature sensor working

guillaumesoucy94, 2025-07-17:
**Describe the issue**
I just setup a RaspberryPi running Ubuntu Server 24.04 LTS and Thingsboard Gateway, it connects successfully with my Thingsboard host, I can see CPU usage, RAM usage, disk space remains and more of the RPi...

I just need to know how to get the CPU temperature to start with and also, and secondly, USB probes temperature readings. (When I will receive them.)

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

This is the gateway configuration file

[tb_gateway.json](https://github.com/user-attachments/files/21277341/tb_gateway.json)

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [Ubuntu 24.04 LTS]
 - Thingsboard IoT Gateway version [3.7.6]
 - Python version [3.12.3]

samson0v, 2025-07-18:
Hi @guillaumesoucy94, thanks for your interest in ThingsBoard IoT Gateway!
Did I understand correctly that in addition to the default statistics, you also need the CPU temperature and USB probes temperature?

guillaumesoucy94, 2025-07-20:
Hello @samson0v 

Yes, I need to get the CPU temp for now, just to test it up as my USB probes aren't arrived yet.

However, once I learn how to get temperature from CPU, as example, it should be easy for me to get readings of my USB probes. I just need to learn the bases of it. 

I know how to get CPU temp by using the command `sensors` in the terminal connecting to the Pi using SSH, this is not a mystery, it is just how-to display that value in TB. 

The story to put in context is that setup will be used to monitor temp in my servers room and if everything goes well, more Raspberry Pi based sensors will be built for fridges/freezers (with external probes) also to monitor temp in different networking closets and other keys spots like emergency generators room etc. 

It's is set to replace WiFi's ones who keeps disconnecting at random times and preventing to know the servers room and our other areas cooling status. 

I appreciate your help and talk to you soon ;-)

guillaumesoucy94, 2025-07-26:
Hello @samson0v 
I don't have a Gateway list so I'd followed [This](https://thingsboard.io/docs/iot-gateway/getting-started/), After importing the first .json file and it creates "ThingsBoard IoT Gateways". When opening the dashboard, the following error is displayed:

Problem loading widget configuration.
Probably associated widget type was removed.

[image]

samson0v, 2025-07-29:
Hi @guillaumesoucy94, you don't need to import the gateway dashboard.
Starting from ThingsBoard v.3.9, you can find the Gateway's management dashboard in the Entities section in the right sidebar:

[image]

Also, we will update the gateway documentation soon.

guillaumesoucy94, 2025-07-31:
After upgrading to 4.1 the WebGUI stopped working even if Ubuntu states that TB is running `service thingsboard status`

`service thingsboard status
● thingsboard.service - thingsboard
     Loaded: loaded (/usr/lib/systemd/system/thingsboard.service; enabled; preset: enabled)
     Active: active (running) since Wed 2025-07-30 19:55:53 EDT; 35s ago
   Main PID: 753 (thingsboard.jar)
      Tasks: 21 (limit: 2268)
     Memory: 331.5M (peak: 331.7M)
        CPU: 16.222s
     CGroup: /system.slice/thingsboard.service
             ├─753 /bin/bash /usr/share/thingsboard/bin/thingsboard.jar
             └─843 /usr/bin/java -Dsun.misc.URLClassPath.disableJarChecking=true -Dplatform=deb -Dinstall.data_dir=/usr/share/thingsboard/data "-Xlog:gc*,heap*,age*,safepoint=debug:file=/var/log/thingsboard/gc.log:time,uptime,level,tags:>

Jul 30 19:55:53 monitoring-dc systemd[1]: Started thingsboard.service - thingsboard.
Jul 30 19:56:19 monitoring-dc thingsboard.jar[843]:   ______    __      _                              ____                               __
Jul 30 19:56:19 monitoring-dc thingsboard.jar[843]:  /_  __/   / /_    (_)   ____    ____ _   _____   / __ )  ____   ____ _   _____  ____/ /
Jul 30 19:56:19 monitoring-dc thingsboard.jar[843]:   / /     / __ \  / /   / __ \  / __ `/  / ___/  / __  | / __ \ / __ `/  / ___/ / __  /
Jul 30 19:56:19 monitoring-dc thingsboard.jar[843]:  / /     / / / / / /   / / / / / /_/ /  (__  )  / /_/ / / /_/ // /_/ /  / /    / /_/ /
Jul 30 19:56:19 monitoring-dc thingsboard.jar[843]: /_/     /_/ /_/ /_/   /_/ /_/  \__, /  /____/  /_____/  \____/ \__,_/  /_/     \__,_/
Jul 30 19:56:19 monitoring-dc thingsboard.jar[843]:                               /____/
Jul 30 19:56:19 monitoring-dc thingsboard.jar[843]:  ===================================================
Jul 30 19:56:19 monitoring-dc thingsboard.jar[843]:  :: ThingsBoard ::       (v3.3.4.1)
Jul 30 19:56:19 monitoring-dc thingsboard.jar[843]:  ===================================================
lines 1-21/21 (END)

service thingsboard status
● thingsboard.service - thingsboard
     Loaded: loaded (/usr/lib/systemd/system/thingsboard.service; enabled; preset: enabled)
     Active: active (running) since Wed 2025-07-30 20:26:15 EDT; 50s ago
   Main PID: 9466 (thingsboard.jar)
      Tasks: 30 (limit: 2268)
     Memory: 468.4M (peak: 468.6M)
        CPU: 48.086s
     CGroup: /system.slice/thingsboard.service
             ├─9466 /bin/bash /usr/share/thingsboard/bin/thingsboard.jar
             └─9481 /usr/bin/java -Dsun.misc.URLClassPath.disableJarChecking=true -Dplatform=deb -Dinstall.data_dir=/usr/share/thingsboard/data "-Xlog:gc*,heap*,age*,safepoint=debug:file=/var/log/thingsboard/gc.log:time,uptime,level,tags>

Jul 30 20:26:15 monitoring-dc thingsboard.jar[9481]: OpenJDK 64-Bit Server VM warning: Option UseBiasedLocking was deprecated in version 15.0 and will likely be removed in a future release.
Jul 30 20:26:22 monitoring-dc thingsboard.jar[9481]:   _____ _     _                 ____                      _
Jul 30 20:26:22 monitoring-dc thingsboard.jar[9481]:  |_   _| |__ (_)_ __   __ _ ___| __ )  ___   __ _ _ __ __| |
Jul 30 20:26:22 monitoring-dc thingsboard.jar[9481]:    | | | '_ \| | '_ \ / _` / __|  _ \ / _ \ / _` | '__/ _` |
Jul 30 20:26:22 monitoring-dc thingsboard.jar[9481]:    | | | | | | | | | | (_| \__ \ |_) | (_) | (_| | | | (_| |
Jul 30 20:26:22 monitoring-dc thingsboard.jar[9481]:    |_| |_| |_|_|_| |_|\__, |___/____/ \___/ \__,_|_|  \__,_|
Jul 30 20:26:22 monitoring-dc thingsboard.jar[9481]:                       |___/
Jul 30 20:26:22 monitoring-dc thingsboard.jar[9481]:  ===========================================================
Jul 30 20:26:22 monitoring-dc thingsboard.jar[9481]:  :: ThingsBoard ::       (v4.1.0)
Jul 30 20:26:22 monitoring-dc thingsboard.jar[9481]:  ===========================================================
lines 1-21/21 (END)

service thingsboard status
● thingsboard.service - thingsboard
     Loaded: loaded (/usr/lib/systemd/system/thingsboard.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2025-07-30 20:27:13 EDT; 13s ago
    Process: 9466 ExecStart=/usr/share/thingsboard/bin/thingsboard.jar (code=exited, status=1/FAILURE)
   Main PID: 9466 (code=exited, status=1/FAILURE)
        CPU: 56.469s

Jul 30 20:27:13 monitoring-dc systemd[1]: thingsboard.service: Consumed 56.469s CPU time, 468.6M memory peak, 0B memory swap peak.`

It starts, then after 90sec, it stops then it restarts and looping like that. 

Guillaume

samson0v, 2025-08-04:
Hi @guillaumesoucy94, I think you should create an issue in the [ThingsBoard](https://github.com/thingsboard/thingsboard) repository for help with the upgrade.

closed 2025-08-16 by guillaumesoucy94