thingsboard/thingsboard-gateway #1901: [BUG] Modbus Serial Connector crashes with uninitialized values

majuss, 2025-07-22:
**Describe the bug**
Clear and concise description of what the bug is.

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

majuss, 2025-07-22:
My hotfix currently is to write a zero to the nan values in simplejson → encoder.py 
line 378

```python
        try:
            print(o)
            replaced = replace_nan(o)
            return _iterencode(replaced, 0)
        finally:
            key_memo.clear()

def replace_nan(obj):
    import math
    if isinstance(obj, float) and math.isnan(obj):
        return 0
    elif isinstance(obj, dict):
        return {k: replace_nan(v) for k, v in obj.items()}
    elif isinstance(obj, list):
        return [replace_nan(elem) for elem in obj]
    else:
        return obj
```

pro100805, 2025-08-07:
The changes helped the connector run smoothly for two days.
I hope it will continue to work without any problems.
👍

pro100805, 2025-08-07:
/var/lib/thingsboard_gateway/venv/lib/python3.11/site-packages/thingsboard_gateway/gateway/tb_gateway_service.py

json_data = dumps(data.to_dict(self.__latency_debug_mode), separators=(',', ':'), skipkeys=True, ignore_nan=True)

gratitude [ishikht](https://github.com/ishikht)

samson0v, 2025-08-15:
Hi @majuss, @ishikht and @pro100805, thanks for your interest in ThingsBoard IoT Gateway!
Issue was fixed via [commit](https://github.com/thingsboard/thingsboard-gateway/commit/ee0c24862284cbc2c7a1b57fd9ae64a418b48166).
Especially thanks to @pro100805 for the solution!
Try to use the latest version of Gateway via the master branch and let us know about the result.

closed 2025-09-04 by samson0v