thingsboard/thingsboard-gateway #1915: [HELP] Unable to connect to an external MQTT broker

FabiocchiF, 2025-08-01:
**Describe the issue**
Greetings;

I'm trying to connect to an external MQTT broker through the IoT gateway component in order to access a production plant data. 
I've setup the gateway entity using thingsboard.demo.io as the host and user+pw as the device credentials.
Then I've created the MQTT connector populating the broker domain, username and pw fields(basic security). For a simple start I've tried to extract just one field from the topic inserted in the *data mapping* tab. I don't know if the fact that the broker uses TLS encryption too could case the problem(it's toggled on on the mqtt explorer page).

The first time I tried to connect I got this error : 
```
tb-gateway  | 2025-08-01 14:42:08.640 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 255 - Remote configuration received, key: ['*my_key*']
tb-gateway  | 2025-08-01 14:42:08.642 - |ERROR| - [tb_logger.py] - tb_logger - exception - 182 - 'MqttConnector' object has no attribute '_client'
tb-gateway  | Traceback (most recent call last):
tb-gateway  |   File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 382, in close
tb-gateway  |     self._client.disconnect()
tb-gateway  |     ^^^^^^^^^^^^
tb-gateway  | AttributeError: 'MqttConnector' object has no attribute '_client'
tb-gateway  | 2025-08-01 14:42:08.644 - |ERROR| - [tb_logger.py] - tb_logger - exception - 182 - Exception on closing connector occurred:
tb-gateway  | Traceback (most recent call last):
tb-gateway  |   File "/thingsboard_gateway/tb_utility/tb_gateway_remote_configurator.py", line 711, in _handle_connector_configuration_update
tb-gateway  |     retrieved_connector.close()
tb-gateway  |   File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 385, in close
tb-gateway  |     self._client.loop_stop()
tb-gateway  |     ^^^^^^^^^^^^
tb-gateway  | AttributeError: 'MqttConnector' object has no attribute '_client'

```

I tried restarting the connection but this time I keep getting this error log, being unable to connect to the tb host itself:
`tb-gateway  | 2025-08-01 14:47:03.654 - |INFO| - [tb_client.py] - tb_client - _on_connect - 314 - MQTT client connected to platform demo.thingsboard.io: 1883
tb-gateway  | 2025-08-01 14:47:03.654 - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 616 - Connection failed with result code 134 (Bad user name or password)
tb-gateway  | 2025-08-01 14:47:03.655 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 591 - MQTT client was disconnected by client with reason code 128 (Unspecified error)`

I've also noticed that the `tb_gateway.json` file in the docker environment and the one in the thingsboard webUI are not exactly matched, and if I try to modify the tb_UI JSON it won't let me do it. I've also tried to update the docker image to no avail.

In hope I've been clear enough thanks for the help.

**Configuration** 

[tb_gateway.txt](https://github.com/user-attachments/files/21554553/tb_gateway.txt)
[docker-compose.txt](https://github.com/user-attachments/files/21554523/docker-compose.txt)

**Connector name (If you need help with some connector/converter):**
MQTT_Valgroup. Can't post the config file as I would like to avoid leaking any data connected to the broker by error. 

**Versions (please complete the following information):**
 - OS: W10 Pro 19045.6093
 - Thingsboard IoT Gateway version 3.7.6
 - Python version 3.9

samson0v, 2025-08-05:
Hi @FabiocchiF, thanks for your interest in ThingsBoard IoT Gateway!
Try to remove this line from docker compose file:
```
- accessToken=myToken
```

FabiocchiF, 2025-08-06:
I've tried removing said line before composing up the docker and while I don't seem to get explicit errors the gateway remains inactive. 

I've got this log upon creating it.
```tb-gateway  | 2025-08-06 08:09:26.083 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - __init__ - 85 - Remote Configurator started
tb-gateway  | 2025-08-06 08:09:26.271 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 962 - Connectors - not found, waiting for remote configuration.
tb-gateway  | 2025-08-06 08:09:26.583 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 255 - Remote configuration received, key: []
tb-gateway  | 2025-08-06 08:09:26.642 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 257 - Gateway connectors initialized.
tb-gateway  | 2025-08-06 08:09:26.643 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 261 - Persistent devices loaded.
tb-gateway  | 2025-08-06 08:09:26.643 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 285 - Gateway started.
```

Then after creating the connector and trying a test topic to subscribe to I've Upgraded the docker (this could be due to my ignorance, I don't know if I have to do it everytime I change some config) and got this log with both the gateway the connector remaining inactive.

```
tb-gateway  | 2025-08-06 08:59:47.481 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 154 - Gateway starting...
tb-gateway  | 2025-08-06 08:59:47.493 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 168 - ThingsBoard IoT gateway version: 3.7.6
tb-gateway  | 2025-08-06 08:59:47.494 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 962 - Connectors - not found, waiting for remote configuration.
tb-gateway  | 2025-08-06 08:59:47.551 - |INFO| - [tb_client.py] - tb_client - connect - 394 - Connecting to ThingsBoard...
tb-gateway  | 2025-08-06 08:59:47.551 - |INFO| - [tb_client.py] - tb_client - __send_connect - 423 - Sending connect to demo.thingsboard.io:1883
tb-gateway  | 2025-08-06 08:59:47.828 - |WARNING| - [tb_client.py] - tb_client - connect - 407 - Cannot use TLS connection on this port. Client will try to connect without TLS.
tb-gateway  | 2025-08-06 08:59:47.828 - |INFO| - [tb_client.py] - tb_client - __send_connect - 423 - Sending connect to demo.thingsboard.io:1883
tb-gateway  | 2025-08-06 08:59:48.109 - |INFO| - [tb_client.py] - tb_client - _on_connect - 314 - MQTT client connected to platform demo.thingsboard.io: 1883
tb-gateway  | 2025-08-06 08:59:48.109 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 600 - MQTT client <paho.mqtt.client.Client object at 0x737cf66da890> - Connected!
tb-gateway  | 2025-08-06 08:59:48.373 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 927 - Service configuration was successfully retrieved and applied.
tb-gateway  | 2025-08-06 08:59:48.373 - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - on_service_configuration - 928 - Current device limits: {'rateLimits': {'messages': '20:1,600:60', 'telemetryMessages': '20:1,600:60', 'telemetryDataPoints': '20:1,600:60'}, 'maxPayloadSize': 65536, 'maxInflightMessages': 100}
tb-gateway  | 2025-08-06 08:59:48.373 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - __on_service_configuration - 345 - Current limits for devices connected through the gateway: {}
tb-gateway  | 2025-08-06 08:59:48.974 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 240 - Gateway core started.
tb-gateway  | 2025-08-06 08:59:48.975 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 267 - Subscribed to *|* with id 1 for device *
tb-gateway  | 2025-08-06 08:59:48.979 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - __init__ - 85 - Remote Configurator started
tb-gateway  | 2025-08-06 08:59:48.980 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 234 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 2.98, 'diskUsage': 0.2, 'freeMemory': 16.2, 'freeDisk': '954.0G', 'gwProcessCpuUsage': 0.0, 'gwMemory': 1.5, 'gwProcessMemoryFullInfo': {'rss': 61308928, 'vms': 1225461760, 'shared': 23986176, 'text': 4096, 'lib': 0, 'data': 174104576, 'dirty': 0, 'uss': 60321792, 'pss': 61018112, 'swap': 0}, 'msgsSentToPlatform': 1, 'msgsReceivedFromPlatform': 0}, 'serviceStats': {'storageMsgPulled': 0, 'storageMsgCount': 0, 'platformMsgPushed': 0, 'platformAttrProduced': 0, 'platformTsProduced': 0}, 'connectorsStats': {}}
tb-gateway  | 2025-08-06 08:59:49.166 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 962 - Connectors - not found, waiting for remote configuration.
tb-gateway  | 2025-08-06 08:59:49.479 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 255 - Remote configuration received, key: []
tb-gateway  | 2025-08-06 08:59:49.542 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 257 - Gateway connectors initialized.
tb-gateway  | 2025-08-06 08:59:49.542 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 261 - Persistent devices loaded.
tb-gateway  | 2025-08-06 08:59:49.542 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 285 - Gateway started.

FabiocchiF, 2025-08-06:
Hello; I've done another try this time deleting the access token field **after** composing the docker and this time I got the gateway active; the problem now is with the connector. I've populated only the `data mapping` field to subscirbe to a topic and tried enabling/disabling it changing every time some parameter. Thanks.

```
tb-gateway  | 2025-08-06 09:53:27.706 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 255 - Remote configuration received, key: ['active_connectors']
tb-gateway  | 2025-08-06 09:53:27.709 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 255 - Remote configuration received, key: ['mqttCOMPANY']
tb-gateway  | 2025-08-06 09:53:27.714 - |INFO| - [mqtt_connector.py] - mqtt_connector - __init__ - 145 - Old MQTT connector configuration format detected. Automatic conversion is applied.
tb-gateway  | 2025-08-06 09:53:27.714 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 305 - Mandatory key 'topicFilter' found in mapping handler: {'converter': {'type': 'json', 'attributes': [], 'timeseries': [{'key': 'Nome', 'type': 'string', 'value': '${Records[0].Value}'}], 'deviceInfo': {'deviceNameExpressionSource': 'topic', 'deviceNameExpression': 'Husky13macchina', 'deviceProfileExpressionSource': 'topic', 'deviceProfileExpression': 'COMPANY'}}, 'topicFilter': 'myCompany/data'}
tb-gateway  | 2025-08-06 09:53:27.714 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 305 - Mandatory key 'converter' found in mapping handler: {'converter': {'type': 'json', 'attributes': [], 'timeseries': [{'key': 'Nome', 'type': 'string', 'value': '${Records[0].Value}'}], 'deviceInfo': {'deviceNameExpressionSource': 'topic', 'deviceNameExpression': 'Husky13macchina', 'deviceProfileExpressionSource': 'topic', 'deviceProfileExpression': 'COMPANY'}}, 'topicFilter': 'myTopic/data'}
tb-gateway  | 2025-08-06 09:53:27.714 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 313 - mapping handler has all mandatory keys => accepted: {'converter': {'type': 'json', 'attributes': [], 'timeseries': [{'key': 'Nome', 'type': 'string', 'value': '${Records[0].Value}'}], 'deviceInfo': {'deviceNameExpressionSource': 'topic', 'deviceNameExpression': 'Husky13macchina', 'deviceProfileExpressionSource': 'topic', 'deviceProfileExpression': 'COMPANY'}}, 'topicFilter': 'myTopic/data'}
tb-gateway  | 2025-08-06 09:53:27.715 - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 316 - Number of accepted mapping handlers: 1
tb-gateway  | 2025-08-06 09:53:27.715 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 320 - Number of rejected mapping handlers: 0
tb-gateway  | 2025-08-06 09:53:27.715 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 293 - 'serverSideRpc' section missing from configuration
tb-gateway  | 2025-08-06 09:53:27.715 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 293 - 'connectRequests' section missing from configuration
tb-gateway  | 2025-08-06 09:53:27.715 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 293 - 'disconnectRequests' section missing from configuration
tb-gateway  | 2025-08-06 09:53:27.715 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 293 - 'attributeRequests' section missing from configuration
tb-gateway  | 2025-08-06 09:53:27.715 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 293 - 'attributeUpdates' section missing from configuration
tb-gateway  | 2025-08-06 09:53:27.716 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - __init__ - 253 - Connector connecting anonymously
tb-gateway  | 2025-08-06 09:53:27.716 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init_and_start_regular_connector - 1043 - Cannot find separated report strategy for connector 'mqttCOMPANY'.                              The main report strategy                              will be used as a connector report strategy.
tb-gateway  | 2025-08-06 09:53:27.778 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 522 - "mqttCOMPANY" was disconnected. (<paho.mqtt.client.Client object at 0x7ac51099b110>, None, <MQTTErrorCode.MQTT_ERR_CONN_LOST: 7>)
tb-gateway  | 2025-08-06 09:53:28.219 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 255 - Remote configuration received, key: ['mqttCOMPANY']
tb-gateway  | 2025-08-06 09:53:28.845 - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 522 - "mqttCOMPANY" was disconnected. (<paho.mqtt.client.Client object at 0x7ac51099b110>, None, <MQTTErrorCode.MQTT_ERR_CONN_LOST: 7>)
```

FabiocchiF, 2025-08-06:
Hello, thanks a lot. I've tried connecting to the mosquitto test broker and the connector worked, so the problem probably lies in the TLS encryption of the plant broker. I'll close the topic once I'll figure this out, I remember reading about encryption key in the guide. Thanks and have a good day!

samson0v, 2025-08-06:
@FabiocchiF thanks, you too!