thingsboard/thingsboard-gateway #1924: Gateway 3.7.6 Cannot send message to platform

pro100805, 2025-08-10:
thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; preset: enabled)
     Active: active (running) since Tue 2025-08-05 21:38:07 CEST; 4 days ago
   Main PID: 6349 (python3)
      Tasks: 21 (limit: 1045)
     Memory: 225.7M
        CPU: 1d 2h 49min 33.647s
     CGroup: /system.slice/thingsboard-gateway.service
             └─6349 /var/lib/thingsboard_gateway/venv/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"

Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 75 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.
Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 79 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.
Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 68 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.
Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 75 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.
Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 80 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.
Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 68 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.
Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 75 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.
Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 80 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.
Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 68 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.
Aug 10 18:44:26 debian python3[6349]: 2025-08-10 18:44:26 - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 75 seconds, queue size exceeded, current max inflight messages: 80, max queued messages: 50000.

pro100805, 2025-08-10:
Disconnecting/connecting connectors had no effect.
Worked normally after restarting the service.

maghibus, 2025-08-19:
Hi, I have the same problem.

```
""2025-08-19 09:32:17" - |INFO| - [bmt_connector.py] - bmt_connector - __convert_data - 839 - Confirmation Sent"
""2025-08-19 09:32:17" - |INFO| - [bmt_connector.py] - bmt_connector - __convert_data - 841 - BMT Connector"
""2025-08-19 09:32:17" - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 41031 seconds, queue size exceeded, current max inflight messages: 0, max queued messages: 0."
""2025-08-19 09:32:17" - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 41030 seconds, queue size exceeded, current max inflight messages: 0, max queued messages: 0."
""2025-08-19 09:32:17" - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 40977 seconds, queue size exceeded, current max inflight messages: 0, max queued messages: 0."
""2025-08-19 09:32:17" - |WARNING| - [tb_device_mqtt.py] - tb_device_mqtt - __send_split_message - 1169 - Cannot send message to platform in 41011 seconds, queue size exceeded, current max inflight messages: 0, max queued messages: 0."
""2025-08-19 09:32:17" - |INFO| - [bmt_connector.py] - bmt_connector - __convert_data - 770 - 438"

```

samson0v, 2025-09-04:
Hi @pro100805, @maghibus, @arthuryunze, thanks for your interest in ThingsBoard IoT Gateway!
Could you write the steps to reproduce, including config files and DEBUG logs?
Also, could you confirm that this bug is reproducible on Gateway 3.7.7?