thingsboard/thingsboard-gateway #1945: [BUG] Modbus Connector - Fail to disconnect devices

zuyan9, 2025-08-26:
**Describe the bug**
The Modbus connector is failing to disconnect devices, when there is a configuration change.

In the log below, we can see that, upon a connector config change, ThingsBoard IoT gateway tries to disconnect from the Modbus devices (aka, servers or slaves), but fails.

**Connector name (If bug in the some connector):**
Modbus Connector, over TCP.

**Error traceback (If available):**
```
2025-08-26 00:20:55 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 234 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 10.07, 'diskUsage': 37.9, 'freeMemory': 24.0, 'freeDisk': '16.6G', 'gwProcessCpuUsage': 27.5, 'gwMemory': 5.9, 'gwProcessMemoryFullInfo': {'rss': 114991104, 'vms': 2866573312, 'shared': 24735744, 'text': 4096, 'lib': 0, 'data': 734416896, 'dirty': 0, 'uss': 115859456, 'pss': 116467712, 'swap': 0}, 'msgsSentToPlatform': 543, 'msgsReceivedFromPlatform': 0}, 'serviceStats': {'storageMsgPulled': 543, 'storageMsgCount': 0, 'platformMsgPushed': 542, 'platformAttrProduced': 0, 'platformTsProduced': 3748}, 'connectorsStats': {'MODBUS_Conn': {'connectorMsgsReceived': 4796, 'connectorBytesReceived': 830877, 'storageMsgPushed': 1086}, 'MODBUS_Conn_converter': {'convertersMsgProcessed': 543, 'convertersAttrProduced': 0, 'convertersTsProduced': 3754}}}
2025-08-26 00:21:09 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 255 - Remote configuration received, key: ['MODBUS_Conn']
2025-08-26 00:21:14 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device22R: 
2025-08-26 00:21:19 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device22F: 
2025-08-26 00:21:24 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device23: 
2025-08-26 00:21:29 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device23L: 
2025-08-26 00:21:34 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device23R: 
2025-08-26 00:21:39 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device23F: 
2025-08-26 00:21:44 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device24: 
2025-08-26 00:21:49 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device24L: 
2025-08-26 00:21:54 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device24R: 
2025-08-26 00:21:55 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 234 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 9.78, 'diskUsage': 37.9, 'freeMemory': 24.1, 'freeDisk': '16.6G', 'gwProcessCpuUsage': 7.1, 'gwMemory': 6.0, 'gwProcessMemoryFullInfo': {'rss': 116867072, 'vms': 2418618368, 'shared': 24821760, 'text': 4096, 'lib': 0, 'data': 290136064, 'dirty': 0, 'uss': 116854784, 'pss': 117463040, 'swap': 0}, 'msgsSentToPlatform': 124, 'msgsReceivedFromPlatform': 2}, 'serviceStats': {'storageMsgPulled': 116, 'storageMsgCount': 3, 'platformMsgPushed': 117, 'platformAttrProduced': 0, 'platformTsProduced': 810}, 'connectorsStats': {'MODBUS_Conn_converter': {'convertersMsgProcessed': 119, 'convertersAttrProduced': 0, 'convertersTsProduced': 822}, 'MODBUS_Conn': {'storageMsgPushed': 238, 'connectorMsgsReceived': 1103, 'connectorBytesReceived': 190777, 'connectorsErrors': 9}}}
2025-08-26 00:21:59 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device24F: 
2025-08-26 00:22:04 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device25: 
2025-08-26 00:22:09 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device25L: 
2025-08-26 00:22:14 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device25R: 
2025-08-26 00:22:19 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device25F: 
2025-08-26 00:22:24 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device26: 
2025-08-26 00:22:29 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device26L: 
2025-08-26 00:22:34 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device26R: 
2025-08-26 00:22:39 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device26F: 
2025-08-26 00:22:44 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device27: 
2025-08-26 00:22:49 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device27L: 
2025-08-26 00:22:54 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device27R: 
2025-08-26 00:22:55 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 234 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 9.9, 'diskUsage': 37.9, 'freeMemory': 24.5, 'freeDisk': '16.6G', 'gwProcessCpuUsage': 1.9, 'gwMemory': 6.0, 'gwProcessMemoryFullInfo': {'rss': 116867072, 'vms': 2418618368, 'shared': 24821760, 'text': 4096, 'lib': 0, 'data': 290136064, 'dirty': 0, 'uss': 116862976, 'pss': 117471232, 'swap': 0}, 'msgsSentToPlatform': 11, 'msgsReceivedFromPlatform': 0}, 'serviceStats': {'storageMsgPulled': 0, 'storageMsgCount': 3, 'platformMsgPushed': 0, 'platformAttrProduced': 0, 'platformTsProduced': 0}, 'connectorsStats': {'MODBUS_Conn': {'connectorsErrors': 12}}}
2025-08-26 00:22:59 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device27F: 
2025-08-26 00:23:04 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device28: 
2025-08-26 00:23:09 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device28L: 
2025-08-26 00:23:14 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device28R: 
2025-08-26 00:23:19 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device28F: 
2025-08-26 00:23:24 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device29: 
2025-08-26 00:23:29 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device29L: 
2025-08-26 00:23:34 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device29R: 
2025-08-26 00:23:39 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device29F: 
2025-08-26 00:23:44 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device30: 
2025-08-26 00:23:49 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device30L: 
2025-08-26 00:23:54 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device30R: 
2025-08-26 00:23:55 - |INFO| - [statistics_service.py] - statistics_service - __send_statistics - 234 - Collected regular statistics: {'machineStats': {'totalCpuUsage': 9.31, 'diskUsage': 37.9, 'freeMemory': 24.4, 'freeDisk': '16.6G', 'gwProcessCpuUsage': 2.0, 'gwMemory': 6.0, 'gwProcessMemoryFullInfo': {'rss': 116867072, 'vms': 2418618368, 'shared': 24821760, 'text': 4096, 'lib': 0, 'data': 290136064, 'dirty': 0, 'uss': 116871168, 'pss': 117479424, 'swap': 0}, 'msgsSentToPlatform': 11, 'msgsReceivedFromPlatform': 0}, 'serviceStats': {'storageMsgPulled': 0, 'storageMsgCount': 3, 'platformMsgPushed': 0, 'platformAttrProduced': 0, 'platformTsProduced': 0}, 'connectorsStats': {'MODBUS_Conn': {'connectorsErrors': 12}}}
2025-08-26 00:23:59 - |ERROR| - [slave.py] - slave - close - 166 - Failed to disconnect from Device30F: 
2025-08-26 00:23:59 - |INFO| - [modbus_connector.py] - modbus_connector - close - 133 - MODBUS_Conn has been stopped
2025-08-26 00:24:00 - |ERROR| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_connector_configuration_update - 717 - Connector fc0d56a0-f1f0-47f7-adeb-de6e97e5b9fd not stopped in 5 seconds
2025-08-26 00:24:00 - |INFO| - [modbus_connector.py] - modbus_connector - __init__ - 95 - Starting Modbus Connector...
2025-08-26 00:24:00 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init_and_start_regular_connector - 1060 - Cannot find separated report strategy for connector 'MODBUS_Conn'.                              The main report strategy                              will be used as a connector report strategy.
2025-08-26 00:24:00 - |INFO| - [slave.py] - slave - connect - 230 - Connected to Device17
2025-08-26 00:24:00 - |INFO| - [slave.py] - slave - connect - 230 - Connected to Device18
[... cut ...] nnectorsStats': {'MODBUS_Conn': {'connectorsErrors': 1, 'connectorMsgsReceived': 2235, 'connectorBytesReceived': 385718, 'storageMsgPushed': 608}, 'MODBUS_Conn_converter': {'convertersMsgProcessed': 304, 'convertersAttrProduced': 0, 'convertersTsProduced': 2128}}}
```

**Versions (please complete the following information):**
 - OS: ThingsBoard Cloud
 - Thingsboard IoT Gateway version: [3.7.7](https://github.com/thingsboard/thingsboard-gateway/tree/3.7.7)
 - Python version: NA, running in docker

zuyan9, 2025-12-09:
Still getting logs like this in version 3.8.0
```
2025-12-09 06:09:04.497 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 255 - Remote configuration received, key: ['deleted']
2025-12-09 06:09:04.498 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _process_config_request - 255 - Remote configuration received, key: ['active_connectors']
2025-12-09 06:09:09.511 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device1L: 
2025-12-09 06:09:14.513 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device1R: 
2025-12-09 06:09:19.515 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device2: 
2025-12-09 06:09:24.516 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device2L: 
2025-12-09 06:09:29.517 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device2R: 
2025-12-09 06:09:34.519 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device3: 
2025-12-09 06:09:39.520 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device3L: 
2025-12-09 06:09:44.522 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device3R: 
2025-12-09 06:09:49.525 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device4: 
2025-12-09 06:09:54.527 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device4L: 
2025-12-09 06:09:59.528 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device4R: 
2025-12-09 06:10:04.531 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device5: 
2025-12-09 06:10:09.532 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device5L: 
2025-12-09 06:10:14.537 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device5R: 
2025-12-09 06:10:19.538 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device6: 
2025-12-09 06:10:24.539 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device6L: 
2025-12-09 06:10:29.541 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device6R: 
2025-12-09 06:10:34.542 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device7: 
2025-12-09 06:10:39.543 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device7L: 
2025-12-09 06:10:44.545 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device7R: 
2025-12-09 06:10:49.547 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device8: 
2025-12-09 06:10:54.548 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device8L: 
2025-12-09 06:10:59.554 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device8R: 
2025-12-09 06:11:04.555 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device9: 
2025-12-09 06:11:09.556 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device9L: 
2025-12-09 06:11:14.559 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device9R: 
2025-12-09 06:11:19.562 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device10: 
2025-12-09 06:11:24.563 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device10L: 
2025-12-09 06:11:29.565 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device10R: 
2025-12-09 06:11:34.566 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device11: 
2025-12-09 06:11:39.568 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device11L: 
2025-12-09 06:11:44.569 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device11R: 
2025-12-09 06:11:49.571 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device12: 
2025-12-09 06:11:54.572 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device12L: 
2025-12-09 06:11:59.576 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device12R: 
2025-12-09 06:12:04.580 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device13: 
2025-12-09 06:12:09.581 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device13L: 
2025-12-09 06:12:14.582 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device13R: 
2025-12-09 06:12:19.583 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device14: 
2025-12-09 06:12:24.584 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device14L: 
2025-12-09 06:12:29.584 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device14R: 
2025-12-09 06:12:34.587 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device15: 
2025-12-09 06:12:39.588 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device15L: 
2025-12-09 06:12:44.590 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device15R: 
2025-12-09 06:12:49.591 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device16: 
2025-12-09 06:12:54.592 - |ERROR| - [slave.py] - slave - close - 165 - Failed to disconnect from Device16L: 
[... cut ...]
2025-12-09 06:13:11.303 - |INFO| - [slave.py] - slave - connect - 228 - Connected to Device14
2025-12-09 06:13:11.482 - |INFO| - [slave.py] - slave - connect - 228 - Connected to Device16F
2025-12-09 06:13:11.576 - |INFO| - [slave.py] - slave - connect - 228 - Connected to Device9R
2025-12-09 06:13:11.664 - |INFO| - [slave.py] - slave - connect - 228 - Connected to Device12R
2025-12-09 06:13:11.864 - |INFO| - [slave.py] - slave - connect - 228 - Connected to Device14L
2025-12-09 06:13:11.976 - |INFO| - [slave.py] - slave - connect - 228 - Connected to Device14R
```

18928795339, 2026-09-23:
I also received this log message in 3.8.1. Did you finally fix the problem。