thingsboard/thingsboard-gateway #1946: [BUG] Modbus Connector - Cannot change reportStrategy in the configuration file

zuyan9, 2025-08-26:
**Describe the bug**
It's possible to set the `reportStrategy` using GUI, but not in the json config file.

**Connector name (If bug in the some connector):**
Modbus

**Error traceback (If available):**
`reportStrategy` gets reset when changed in Advanced - Configuration.

**Versions (please complete the following information):**
- OS: ThingsBoard Cloud
- Thingsboard IoT Gateway version: [3.7.7](https://github.com/thingsboard/thingsboard-gateway/tree/3.7.7)
- Python version: NA, running in docker

imbeacon, 2025-08-26:
Hi @zuyan9,

You can change reports strategy in General Tab. Fields that set on the highest level will be overwritten by UI. They present in the configuration in persistence purposes. If you want to change these fields directly - you can change them in related shared attribute.

zuyan9, 2025-08-26:
> Hi [@zuyan9](https://github.com/zuyan9),
>
> You can change reports strategy in General Tab. Fields that set on the highest level will be overwritten by UI. They present in the configuration in [...]

For example, we have this config:
```
{
  "master": {
    "slaves": [...]
  },
  "slave": {},
  "name": "MODBUS_conn",
  "reportStrategy": {
    "type": "ON_CHANGE_OR_REPORT_PERIOD",
    "reportPeriod": 60000
  },
  "logLevel": "ERROR",
  "enableRemoteLogging": true,
  "id": "xxxxx",
  "configVersion": "3.7.7"
}
```

This has the top-level `reportStrategy` config. But in this file, we should not edit the `reportStrategy`? We can only change the top-level `reportStrategy` in GUI? Can we change the device-level `reportStrategy` or data-type-level `reportStrategy` in the config file? How can we change this in the related shared attribute too?

closed 2025-09-04 by zuyan9